burzumishi/linux-baytrail-flexx10 #1: Cannot load the kernel

srikantpatnaik, 2016-01-06:
Hello, nice documentation. Sorry, I have no other means to contact you, hence this issue. 
I have tried to replicate your findings but failed, is it possible for you to upload your kernel/initrd deb files?
I'm desperate to see audio working. Thanks. 
You may email me at srikant@fossee.in, and remove this discussion.

burzumishi, 2016-01-21:
Hello, thank you!

I'm completing this documentation as I advance with my device.

I've not been able to enable sound successfully yet, I only get some crackles but no working sound.

Audio device is not being listed with lspci.

Updating to kernel 4.4 and latest intel sound firmware may help.

I will update "asound.state" file and kernel ".config" as soon as possible.

I guess we will get a working sound soon.

srikantpatnaik, 2016-01-21:
I also tried various settings, but none worked. I'm following this page closely http://ubuntuforums.org/showthread.php?t=2254322&page=53.

burzumishi, 2016-01-21:
Thank you, I have followed that forum too ;)

Some ASUS laptops/tablets has the same ASoC audio chipset and load the same modules but this machines can show the audio hardware with lspci :? 

I've checked the modules loaded in one of those asus laptops and they are the same that we can see loaded in a flexx10 device:

```
snd_soc_sst_bytcr_rt5640    20480  4
snd_soc_rt5640        118784  2 snd_soc_sst_bytcr_rt5640
snd_intel_sst_acpi     16384  2
snd_soc_rl6231         16384  1 snd_soc_rt5640
snd_soc_sst_acpi       16384  1 snd_intel_sst_acpi
snd_intel_sst_core     73728  1 snd_intel_sst_acpi
snd_soc_sst_mfld_platform    90112  3 snd_intel_sst_core
snd_soc_core          212992  3 snd_soc_rt5640,snd_soc_sst_mfld_platform,snd_soc_sst_bytcr_rt5640
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm_oss            49152  0
snd_mixer_oss          24576  1 snd_pcm_oss
snd_pcm               102400  7 snd_soc_rt5640,snd_pcm_oss,snd_soc_core,snd_soc_sst_mfld_platform,snd_soc_sst_bytcr_rt5640,snd_pcm_dmaengine
snd_timer              32768  2 snd_pcm
snd                    81920  13 snd_pcm_oss,snd_soc_core,snd_timer,snd_pcm,snd_soc_sst_mfld_platform,snd_compress,snd_mixer_oss
soundcore              16384  1 snd
ac97_bus               16384  1 snd_soc_core
```

I think it's not working because something is wrong with firmware files or I have not found the good ones, I'm using the latest downloaded from git.

I have built a new kernel and tried some new options to define ALSA values.

I have updated repo with:
```
- New kernel .config for latest git version 4.4.0-rc8+.
- New asound.state 
- An script to configure ALSA values.
- More files ;)
```

Now I got more crackles at boot and login time, I can see the audio device in alsamixer and gnome to change values but still no sound at all. ;(

#### dmesg output:
```
[   12.997067] snd_soc_sst_acpi: module license 'unspecified' taints kernel.
[   12.997070] snd_soc_sst_acpi: module license 'unspecified' taints kernel.
[   12.997076] Disabling lock debugging due to kernel taint
[   13.321130] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[   13.321262] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[   13.321306] compress asoc: snd-soc-dummy-dai <-> compress-cpu-dai mapping ok
[   13.343760] bytcr_rt5640 bytcr_rt5640: rt5640-aif1 <-> ssp2-port mapping ok
[   13.723826] sst-mfld-platform sst-mfld-platform: Slot control: codec_out tx interleaver slot 0 doesn't have DAPM widget!!!
[   13.723908] sst-mfld-platform sst-mfld-platform: Slot control: codec_out tx interleaver slot 1 doesn't have DAPM widget!!!
[   13.723986] sst-mfld-platform sst-mfld-platform: Slot control: codec_out tx interleaver slot 2 doesn't have DAPM widget!!!
[   13.724081] sst-mfld-platform sst-mfld-platform: Slot control: codec_out tx interleaver slot 3 doesn't have DAPM widget!!!
[   13.724148] sst-mfld-platform sst-mfld-platform: Slot control: codec_in rx deinterleaver codec_in0_0 doesn't have DAPM widget!!!
[   13.724206] sst-mfld-platform sst-mfld-platform: Slot control: codec_in rx deinterleaver codec_in0_1 doesn't have DAPM widget!!!
[   13.724260] sst-mfld-platform sst-mfld-platform: Slot control: codec_in rx deinterleaver codec_in1_0 doesn't have DAPM widget!!!
[   13.724318] sst-mfld-platform sst-mfld-platform: Slot control: codec_in rx deinterleaver codec_in1_1 doesn't have DAPM widget!!!
[   18.756276] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   18.756302] intel_sst_acpi 80860F28:00: fw returned err -16
[   19.755676] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   19.755686] intel_sst_acpi 80860F28:00: fw returned err -16
[   20.553122]  Baytrail Audio Port: ASoC: no backend DAIs enabled for Baytrail Audio Port
[   20.553447]  Baytrail Audio Port: ASoC: no backend DAIs enabled for Baytrail Audio Port
[   20.553605]  Baytrail Audio Port: ASoC: no backend DAIs enabled for Baytrail Audio Port
[   21.572544] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   21.572570] intel_sst_acpi 80860F28:00: fw returned err -16
[   21.573042] sst-mfld-platform sst-mfld-platform: Enter: enable=0 port_name=ssp2-port
[   22.572546] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   22.572571] intel_sst_acpi 80860F28:00: fw returned err -16
[   23.572532] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   23.572558] intel_sst_acpi 80860F28:00: fw returned err -16
[   24.572002] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   24.572014] intel_sst_acpi 80860F28:00: fw returned err -16
[   24.572256] sst-mfld-platform sst-mfld-platform: Enter: enable=0 port_name=ssp2-port
[   25.572582] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   25.572607] intel_sst_acpi 80860F28:00: fw returned err -16
[   26.572469] intel_sst_acpi 80860F28:00: Wait timed-out condition:0x0, msg_id:0x1 fw_state 0x3
[   26.572495] intel_sst_acpi 80860F28:00: fw returned err -16
[ 1692.447220] bytcr_rt5640 bytcr_rt5640: BUG: , pos = -22, buffer size = 203830, period size = 1199
[ 1692.447239]  Baytrail Audio Port: sst: error code = -22
[ 1692.447245] bytcr_rt5640 bytcr_rt5640: BUG: , pos = -22, buffer size = 203830, period size = 1199
[... cut ...]
[ 1695.730460] bytcr_rt5640 bytcr_rt5640: BUG: , pos = -22, buffer size = 203830, period size = 1199
[ 1695.730611]  Baytrail Audio Port: sst: error code = -22
[ 1695.730622]  Baytrail Audio Port: sst: error code = -22
[ 1695.730651]  Baytrail Audio Port: sst: error code = -22
[ 1695.734145]  Baytrail Audio Port: sst: error code = -22
[ 1695.734165]  Baytrail Audio Port: sst: error code = -22
[ 1695.734205]  Baytrail Audio Port: sst: error code = -22
[ 1695.734448]  Baytrail Audio Port: sst: error code = -22
[ 1695.734475]  Baytrail Audio Port: sst: error code = -22
```

burzumishi, 2016-08-17:
I'm closing this issue because kernel is loading with many distributions.

closed 2016-08-17 by burzumishi