burzumishi/linux-baytrail-flexx10 #3: The resulting kernel is huge! ~ 2.8G

nicman23, 2016-03-09:
We should probably try to slim the .config .... Will try to do this next week.

Basic plan: 1.remove radeonsi, noveau (i flanked french - no idea how new is written)
2.remove non intel powersaving / mobo management / chipset stuff
3.remove any network cards (pci)
4.Hope did nothing to kill everything

Will make a new pull request :)

burzumishi, 2016-03-09:
Yes you are right! Thank you! 

I've been working on removing pci network cards, non intel chipsets and other stuff.

I'll update a new ".config" using the latest kernel sources, I guess to do it today (I've been a bit busy these week to play with my device). :)

nicman23, 2016-03-10:
Nice :), btw i wont be able to test anything until after the weekend (irl stuff + do not have my charger :P )

A bit not relevant: there multiple clone of the device we are using. Such as vero w10i, odys wintab 10 and of course flexx10.

I am currently using a vero flashed with odys uefi/bios and should not have too much of an impact... Except maybe the audio card which might be different. You know of any way to get the model?

burzumishi, 2016-03-14:
Yes you are right, the sound card codec loaded by the kernel is **rt5640** but I've have not any advance on that point, all required sound modules are loaded but I can't see the device listed with "**lspci**".

I've uploaded a configuration script stored in "**sound**" directory.

nicman23, 2016-03-14:
you would not see it through lspci. It is through sdio and acpi iirc (an interface through sd cards) so it is not a pci device.

emelcher, 2016-04-24:
To reduce kernel size try:
`make INSTALL_MOD_STRIP=1 -j 4 deb-pkg`

burzumishi, 2016-04-25:
I'm using "make deb-pkg" instead of "make; make install", this is stripping my kernel build on Debian.

nicman23, 2016-07-08:
Closing - this is irrelevant as it is the users' concern. He or she should edit the default - distro provided - config to add the needed codecs and drivers for the sound.

closed 2016-07-08 by nicman23