burzumishi/linux-baytrail-flexx10 #6: We should probably warn users about the audio with big giant letters....

nicman23, 2016-04-06:
So my left speaker died.... The only sound it (with windows as alsa still will not work) produces is the one might get by scratching steel.

I understand you do document the audio thing in the readme but should either remove the asound file (which does not work anyways) and document the procedure of configuring alsa or up the character size in the warning.

What are your experience with sound, with the tablet? Also what do you think we should do? - nothing is a valid course of action.

burzumishi, 2016-04-10:
Oh I'm so sorry! My speakers are working on windows.

While booting debian I get a lot of FW errors but I can hear some crakles and the boot process is slower.

After login to desktop I can hear some crackles again and the system get frozen for some seconds while loading the sound mixer... then I can use the desktop but if any program has sound it gets frozen for a while.

The problem seems to be with the FW files, I have tried some files but I get FW errors always ... the asound.state file was copied from /var/lib/alsa/

If you're only deleting the asound.state file it will be written again at shutdown or reboot.
```

rm /var/lib/alsa/asound.state # remove the state file
chmod -x /usr/sbin/alsactl # make alsactl non-executable to prevent settings being written on shutdown
<REBOOT>
chmod +x /usr/sbin/alsactl

```
Have you tried this?

nicman23, 2016-07-08:
Closing as fixed. I will update details and do a pull request.

closed 2016-07-08 by nicman23