burzumishi/linux-baytrail-flexx10 #7: Possible progress on the audio

nicman23, 2016-04-26:
There is some progress/traction on the asus x205ta that has the same uncommon audio card.

the bug is tracked [here](https://bugzilla.kernel.org/show_bug.cgi?id=117141).

Our is the insyde reference - see description.

PS: do not try anything from other models. It will not work. Our chip does not have 2 pins (or something technical like that) and it  will timeout alsa. Basically, alsa, cannot start the chip - let alone produce sound. No amount of hacking/ trying different firmwares will do anything.

tl;dr : follow the link ; pray to Saint Gnusious

nicman23, 2016-05-05:
If anyone cares to try a diff from a intel dude's repo (that i have forked to make a pull request to get his changes to a more recent repo) you can use that [link](https://patch-diff.githubusercontent.com/raw/nicman23/sound/pull/2.diff) :)

burzumishi, 2016-05-06:
Thank you! It looks Good! I'll give it a try this weekend! :)

nicman23, 2016-05-13:
Did you try it @burzumishi ? Mine is currently on rma hell :C . Also could you post your dmidecode in the bugtracker?

nicman23, 2016-06-23:
can you give me your dmidecode ? I just made an account and we should add it for our machines to be added in the quirk table. Also I will be receiving my tablet from rma tomorrow and i ll run some tests :)

burzumishi, 2016-06-27:
Hi! dmidecode flexx10: https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/doc/dmidecode.out

nicman23, 2016-06-27:
Ok I will upload it in the bugtracker when I go home :)

burzumishi, 2016-06-27:
OK thank you! We will have to wait for the patch update :)

nicman23, 2016-06-28:
Uploaded, also need recompiled dsdt tablet, as per div request :-)

nicman23, 2016-06-29:
I'm close... The problem is that the irqs in the dsdt are not how the kernel expects them (as per http://mailman.alsa-project.org/pipermail/alsa-devel/2015-July/094627.html)...

I edited my dsdt and recompiled it... Now there are not dmesg firmware errors. However, there is also no sound... But, I am close...

More in a few days, got some finals left to write. 

ps: i promise a tutorial :)

jingber, 2016-07-03:
I've adapted the patch linked in the second post.  I can load the kernel modules without errors, and hear all kinds of beeps and bonks by playing with the the 1000+ mixer controls, but unfortunately no real audio.

I have a Flexx 9, which is 95% similar.  I can tell you that I need to keep acpi_ipc_irq_index at 0, not at 5.

nicman23, 2016-07-03:
I did not use the kernel patch. I edited the DSDT and loaded it with grub.

There are no clicks or any heat from the speakers. There is also no sound. Please upload a DSDT.dsl file

jingber, 2016-07-03:
Github reports "unfortunately, we don't support that file type".

You'll have to leach it from my web site: http://home.ix.netcom.com/~jjingber/flexx9.dsl

jingber, 2016-07-03:
No clicks or heat here - but if you play some audio file in a loop, and gradually unmute the mixer channels, you're going to get something that sounds like a hearing test.

nicman23, 2016-07-03:
The kernel expects this order: (Basically hex from smaller to bigger)
```
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x00000018,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x00000019,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001A,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001B,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001C,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001D,
                }
```
Your dsdt file has this: (The first one is 1D and then 18, 19 ...)
```
Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001D,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x00000018,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x00000019,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001A,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001B,
                }
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x0000001C,
                }
```

nicman23, 2016-07-03:
Sorry missclick

jingber, 2016-07-03:
<i>Your dsdt file has this: (The first one is 1D and then 18, 19 ...)</i>

Roger.....   Playing with this now.

nicman23, 2016-07-03:
The sound you are hearing is basically dc current in your speakers. I have already blown one and had to rma it (and pay for the replacement) ...

jingber, 2016-07-03:
<i>The sound you are hearing is basically dc current in your speakers. I have already blown one and had to rma it (and pay for the replacement) ...</I>

Understood.

jingber, 2016-07-07:
Alright, got it working  --

1.  Kernel 4.6.3.
2.  Didn't change my DSDT - the acpi_ipc_irq_index at 0 in sound/soc/intel/atom/sst/sst_acpi.c in sufficient.
3.  Use firmware fw_sst_0f28_sp0.bin linked to fw_sst_of28.bin (073bb387eb96f8c01d039778086484ac)
4. Run alsa_cr.sh, followed by alsa_cr_spk.sh from the following archive: https://groups.google.com/group/android-x86/attach/d1ffc290d6d56/alsa_bytcr_andx86.zip?part=0.1&view=1

This is was a big help in getting it going: https://groups.google.com/forum/#!topic/android-x86/un74okmr30g.  Seems the problem on my end was the wrong firmware.

jingber, 2016-07-07:
And DAC1 in the mixer is the master volume control.

jingber, 2016-07-08:
UPDATED: Clarified the instructions and included a missing patch.

nicman23, 2016-07-08:
Yay :) So you did not bother with the dsdt fix? Probably more static that way... I ll try this against arch's kernel and report back (got limited cpu cycles atm)

> https://groups.google.com/forum/#!topic/android-x86/un74okmr30g

Shit I should had added it here sooner, that is where I got most of my info... (that and maillists.... so many mails and threads)

About your 1 and 2, how did you force use spp0 and how did you unload the old old snd_* modules (probably blacklisted them?)

nicman23, 2016-07-08:
also, this patch is static about irq, right? I cannot seem to find any quirk tables...

jingber, 2016-07-08:
1.  No DSDT fix, although that certainly would have obviated the need for the IRQ fix in the source.  The IRQ fix in the source is only a one-liner.

2.  I just removed *all* my old snd_* modules, and rebuilt the kernel modules with just the CONFIG options specified above.  It won't build a duplicate set unless you select both x5640 variants.  You can also certainly blacklist.

3.  It addresses the IRQ mapping, but also quirks the Flexx 9 and 10 in the same fashion as the Dell Venue 8 (grep YIFANG over the patch).

jingber, 2016-07-08:
The following file set will (allegedly) enable automatic headphone/speaker switching through Pulse Audio (untested by me).  http://home.ix.netcom.com/~jjingber/alsa-ucm-5640.tar.gz

nicman23, 2016-07-08:
I cannot apply the patch you listed before ... Getting a 

`patch: **** malformed patch at line 537: +                },`

jingber, 2016-07-08:
Let me look, but try running it through dos2unix - I bounced it around a bunch of machines.

jingber, 2016-07-08:
It's missing a bracket --- gimme a sec...

jingber, 2016-07-08:
Alright.  D/L it again.

nicman23, 2016-07-08:
I fixed my self :) thanks anyways (i do not have a flexx but a clone and I had to make my own quirk entry anyways)

nicman23, 2016-07-08:
DAMN IT WORKS :D

jingber, 2016-07-08:
Damn right!!!   Hehe...

nicman23, 2016-07-08:
there is also this that we might want to look at https://github.com/plbossart/UCM/tree/master/bytcr-rt5640

nicman23, 2016-07-08:
I - with great happiness - closing this bitch of a bug :D

jingber, 2016-07-09:
But now nobody can find it.

nicman23, 2016-07-10:
I will update the readme do not worry ;) , I will create a more general version (adding your device to the quirk table and one for flexx10) and I will make a pull request. Bugtrackers are not for permanent tutorials :P

nicman23, 2016-08-11:
btw @jingber couldn't fw_sst_0f28 in patch be renamed to fw_sst_0f28_sp0 to be a more cleaner install?

nicman23, 2016-08-11:
linux 4.7 did not want to compile :S no error no nothing.

Patches were generated again to allow for a clean environment and they did succeed....

Trying now linux-zen 4.6.5

nicman23, 2016-08-11:
Sound works with zen 4.6.5 and sp0 firmware.

Made pkgbuilds for both on https://github.com/nicman23/misc_aur

nicman23, 2016-08-11:
made a sort of tutorial for arch https://www.reddit.com/r/archlinux/comments/4x8txv/tutorial_intel_atom_tablets_with_bytcr_sound_and/

jingber, 2016-08-12:
[QUOTE]btw @jingber couldn't fw_sst_0f28 in patch be renamed to fw_sst_0f28_sp0 to be a more cleaner install?[/QUOTE]

Sure can, but I don't know the reason for the various file naming so I left it like it is.

jingber, 2016-08-12:
Here is an archive with the patch set I use for Baytrail on 4.8-rc1 (includes a refactored audio patch):

http://home.ix.netcom.com/~jjingber/linux-4.8-baytrail_patchset.tar.gz

nicman23, 2016-08-12:
Look if it is only one line and the alternative is risking a blown speaker / amp, I'll take the one extra line ;)

jingber, 2016-08-17:
The audio patch landed in linux-next and it'll be in 4.8-rc3.  We'll still need to add the two blocks dealing with the Flexx tablets.

burzumishi, 2016-08-17:
Hi, there! 

Great job guys! Congrats!

I'm glad of reading that sound is working on these devices, I've been on holidays and I'm a bit outdated.

I'll test the new sound and Wifi/BT patches with kernel 4.8 and update this github repository, I'll test UCM files for Pulseaudio too.

I'll try to make a newer and smaller kernel configuration without unneeded modules for this devices (eth, scsi, ...) and I'll add the newest AXP20X modules.

Once I've tested all the changes, patches and I've mixed all together ... then I'll update the main README to have a new guide.

Great job!

nicman23, 2016-08-17:
Should we move further developments in a new issue? 

Something like: UCM support, upstream news and automatic firmware picking

burzumishi, 2016-08-17:
I agree, sound is working here now! 

I have just got sound to work with kernel 4.8.0-rc2 merging all patches manually and using the new firmware file.

I'm going to update the repo with the kernel config and full patch files,  including all latest commits done to kernel git repos before applying my changes to the sources. Also,  I have merged the wifi module sources as a staging driver in this patch.

I have installed UCM, I have Mic settings but seems not to be working, needs more testing ...

Tomorrow I'll test your DKMS ... ;)

jingber, 2016-08-18:
Before you guys play with the UCM files, I would suggest waiting for 4.8-rc3 - it incorporates the audio fixes our tablets need, but the patch also is more extensive than what we had to work with.  This should only be a few more days.

burzumishi, 2016-08-18:
Right thanks, I just make sound to work and installed UCM files because I had a kernel without sound yet,  but I didn't anything with UCM ... I'm waiting for RC3 and review the new required changes/patches to get all working again :)

Meanwhile I'll be moving all documentation from the main README to github Wiki pages . :)

nicman23, 2016-08-22:
so @jingber what is it need now? do i have to add my own device to the quirk table?

jingber, 2016-08-23:
I'm working on a patch for -rc3.  It seems it needs a little more than just the quirk.

jingber, 2016-08-29:
I just had a look at this, and the new audio stuff got backed out from -rc3.  

So, you can use the same patch we've been using on -rc3, if you want to run this kernel:  http://home.ix.netcom.com/~jingber/linux-4.8-baytrail_patchset.tar.gz

I'm going to get the new audio stuff out of the kernel git and make a new patch.

jingber, 2016-09-03:
Sorry I took so long on this --

Here is the 'new' audio code from linux-next backported to 4.8-rc4.  Audio works, but I haven't played around any further.

http://home.ix.netcom.com/~jjingber/0013-rt5640_audio_on_intel_baytrail.patch

nicman23, 2016-09-04:
did you still get the distortion on one speaker?

nicman23, 2016-09-04:
also what is the situation with the firmware? do you still have to replace the already-in-upstream with the sp0 one?

jingber, 2016-09-04:
I don't know what caused the distortion, but it happened after a Debian update of pulse audio, and after I restored my original asound.state, it never re-occured.

You still need the symlink to the ssp0 firmware.

nicman23, 2016-09-04:
Btw does linux-next have our devices?

nicman23, 2016-09-04:
also compile exits with 
`sound/soc/intel/skylake/skl-topology.c:24:39: fatal error: uapi/sound/snd_sst_tokens.h: No such file or directory`

jingber, 2016-09-04:
This is the audio branch from linux-next, backported to 4.8-rc4:

http://home.ix.netcom.com/%7Ejjingber/0013-sound_asoc_for_next_20160825.diff

This is the patch to add the quirks for the Flexx 9 and 10 tablets:

http://home.ix.netcom.com/%7Ejjingber/0014-rt5640_audio_on_nextbook_tablets.diff

If you apply both of these to 4.8-rc, you have the the new audio code from linux-next, along with the quirks for our tablets.  I only compile with our audio devices selected.  Being this code is from linux-next, if you're compiling using a generic .config, there is a change something is broken if you're trying to compile the kernel with all of the audio devices selected.

nicman23, 2016-09-05:
Yeah sorry missread a codec - that i thought bytcr needed - and concluded that skylake snd was selected by it :P

[132 more replies]

closed 2017-03-18 by nicman23