burzumishi/linux-baytrail-flexx10 #8: Anyone manage to make KMS work

jingber, 2016-06-29:
Does kernel modesetting work, and if so what did you do to achieve this?  Running Debian Stretch w/kernel 4.5.7 and the 5 patches for wireless, and the three recommended "intel" patches for i915.

So far, I can only run X over intelfb :(.

jingber, 2016-07-08:
Beuler???

Anyone here running with KMS on these tablets?

burzumishi, 2016-08-17:
Right, it's working for me.

glxinfo reports:
`
root@flexx10:/home/burzum-ishi# glxinfo | grep -i "direct render"
direct rendering: Yes
`
Modules loaded:

`
root@flexx10:/home/burzum-ishi# lsmod | grep -i -e drm -e i915 -e kms 
i915                 1208320  4
i2c_algo_bit           16384  1 i915
drm_kms_helper        143360  1 i915
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   364544  6 i915,drm_kms_helper
button                 16384  1 i915
video                  40960  1 i915
`

xrandr report:

`
burzum-ishi@flexx10:~$ xrandr
Screen 0: minimum 8 x 8, current 1280 x 800, maximum 32767 x 32767
DSI1 connected primary 1280x800+0+0 right (normal left inverted right x axis y axis) 0mm x 0mm
   800x1280      59.99*+
DP1 disconnected (normal left inverted right x axis y axis)
DP2 disconnected (normal left inverted right x axis y axis)
HDMI1 disconnected (normal left inverted right x axis y axis)
HDMI2 disconnected (normal left inverted right x axis y axis)
VIRTUAL1 disconnected (normal left inverted right x axis y axis)
`

I'm attaching dmesg with drm with debug enabled for driver and kms:

[dmesg.drm.2.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/423042/dmesg.drm.2.txt)
[dmesg.drm.1.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/423043/dmesg.drm.1.txt)

Xorg log:
[Xorg.0.log.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/423051/Xorg.0.log.txt)

I have blacklisted all FB modules (including intelfb) except vesafb. As I know, this is the default configuration on debian based distros.

I hope this helps you to figure out what's going on there, let me know if you need more information.

nicman23, 2016-08-17:
ditto

[dmesg.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/423442/dmesg.txt)

[Xorg.0.log.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/423444/Xorg.0.log.txt)

jingber, 2016-08-18:
Thanks a bunch.  Those logs are gold.  Will be combing over them and the kernel config.

jingber, 2016-08-18:
Which one of you has the Flexx?  nicman23's panel is eDP while burzumishi's is MIPI DSI.

burzumishi, 2016-08-18:
I have the Flexx10.1 one:

```
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: YIFANG
	Product Name: NXW101QC232
	Version: Type1 - TBD by OEM
	Wake-up Type: Power Switch
	SKU Number: Type1 - SKU0
	Family: M1015BFP T11220300
```

I have pushed all info files from my device to the repo, dmidecode, DSDT, acpidump, etc ...

nicman23, 2016-08-18:
again. i do not have a flexx but a rebrand :)

the only differences is that and that the touchscreen is another model

jingber, 2016-08-20:
Alright, I just made modesetting work for the *first* time on this tablet --

<i>[    1.092476] [drm:intel_opregion_setup] Found valid VBT in ACPI OpRegion (Mailbox #4)
</I>  << -- this is key (from nicman23's log), and where my tablet falls on it's ass.

I'll cut to the chase right here -- this tablet (Flexx 9) doesn't have a full video BIOS and will probably never work 'out of the box' without the following work around.  There's a 2K blob of "something" where at least the Venue 8 (and presumably your tablet) has a full Intel reference Valley View video BIOS (verified by copyright notice in header, from the Dell 8 tablet).  The BIOS is signed with a key that Intel has used to sign other drivers (same as on an Optiplex I have close-by).  The hardware and/or driver don't care about the signature (on Linux, at least).

This is really a very dirty work-around, and I'll do a proper write-up (offtopic for Flexx 10, it appears), but the only solution that I can make work is to have grub load a (modified *) copy of the reference Intel VV/Baytrail video BIOS with the 'loadbios' command prior to booting the kernel.  Further, the i915 module needs to be patched to acquire the location of the VBIOS by scanning kmem for the BIOS signature, then changing the expected location of the BIOS in i915 before it finally bootstraps the hardware.

* The header of the VBIOS contains the boiler-plate Intel binary license which prohibits modification, but I'm hoping it's better to ask for forgiveness for the sake of compatibility and a bug report.  Report going to i915 @ freedesktop.

jingber, 2016-08-20:
<i>IntelGraphicsMem</i>

First 16 bytes of proper VBT.  Flexx 9 has garbage.  So, i915 is looking in the wrong place, or this tablet has reduced (or junk) video routines.

jingber, 2016-08-20:
Modified Intel Valley View (Baytrail) reference VBIOS (re-worked relocation table for Grub load):  http://home.ix.netcom.com/~jjingber/vbios.bin

jingber, 2016-08-20:
FYI -- That's 32-bit firmware.

nicman23, 2016-09-21:
does the 32 bit blob work with amd64 kernels?

jharrison022, 2016-09-21:
Yes.  It's a 32-bit VBIOS (because the UEFI is 32-bit).  I have to re-up it to my site when I get a change - the link seems to go to a zero-byte file.

nicman23, 2016-09-21:
figured as much but it does not hurt to ask :) anyways, I m going to add it to the repo and write some documentation for it, when I ll have the chance

RussianNeuroMancer, 2017-03-17:
@jharrison022 @burzumishi is there no changes with [intel-drm-next](kernel.ubuntu.com/~kernel-ppa/mainline/drm-intel-next/) or 4.11rc2?

nicman23, 2018-01-27:
closing as it is probably fixed?

RussianNeuroMancer, 2018-01-28:
There is still some exceptions, like this: https://bugs.freedesktop.org/show_bug.cgi?id=104705
But most of hardware should work.