burzumishi/linux-baytrail-flexx10 #9: Bricked?

ircncl, 2016-07-25:
I went through with a debian install image, got up to the grub stage and thought it would let me manually fix up anything after that step by letting me jump to one of the VTs, but no, it thought it'd be a good idea to automatically reboot.
In the partition scheme, I manually specified to leave the EFI and 16gb windows partition alone, so I thought it would just be a mild impedance and I'd have to go through the whole install process again, but no, again.
Now I'm greeted with a nice blue screen stating the following:

- - - - - - 8< - - - - - - 
Recovery

Your PC/Device needs to be repaired

A required device isn't connected or can't be accessed.

Error code: 0xc0000225

You'll need to use recovery tools. if you don't have any installation media (like a disc or USB device), contact your PC administrator or PC/Device manufacturer.

Press the Windows key for UEFI Firmware Settings

- - - - - - >8 - - - - - - 

Pressing esc on boot doesn't give me boot options, f2 nor Fn+f2 allow me to get to UEFI settings, and pressing the windows key (as stated in the error message above), on the keyboard or on the back of the device, does not do anything.
Is there anything I can do from here?
Thanks

ircncl, 2016-07-25:
In desperation, I rapidly tapped on esc and f2 while booting...and I got an EFI menu.
rapidly tapping or holding either f2 or esc alone do not bring it up. I have no idea why this is like this, but I've gone through the installation again, and I have a bootable debian environment on this device. Thank you for this guide.

Also: adding "video=efifb fbcon=rotate:1" to the boot options make using the console a lot easier.

closed 2016-07-25 by ircncl

burzumishi, 2016-08-10:
Hi! 

You're welcome, I was on holidays and could not see these messages before.

Holding "Power+VolumeUp" buttons about 5 seconds should boot up your device with an EFI menu.

Hope you're doing well now.