burzumishi/linux-baytrail-flexx10 #11: Battery is not supported

nicman23, 2016-08-11:
but there is progress: https://bugzilla.kernel.org/show_bug.cgi?id=88471

jingber, 2016-08-12:
I have a Flexx 9, not a 10, so I don't know if the instructions on this page are generic or not or from some similar device, but the HW uses the axp288_* kernel modules for hardware monitoring, the maa8425 for spatial positioning and the jsa1212 for ambient light sensing.  The camera on the Flexx 9 is the OVTI2680 which may be different between these models, but I'm willing to bet the other sensors are the same.

The kernel has support for all this hardware with the exception of the camera, but like the LKML link indicates, they don't work properly yet on our hardware.

nicman23, 2016-08-12:
For me it just says that there is no battery, when I cat anything from /sys/class/

jingber, 2016-08-15:
[QUOTE]https://github.com/Icenowy/axpd[/QUOTE]

This workaround for battery status works for me (tm)...

jingber, 2016-08-15:
Some more good news -- enable all the AXP2* modules in the kernel, and now Bluetooth works, at least on 4.8.

jingber, 2016-08-15:
Specifically, the modules in question seem to be axp20x and axp20x_i2c.  With these loaded, the start_bt.sh script fires up the Bluetooth and see my phone.

Hint and credit go to a post on the rtl8723bs_bt forum.

nicman23, 2016-08-15:
Nice. We just need better wifi performance and s3 and lower state and it will finally be usable. 

Btw did you ever get pulse to work without the scripts that state alsa?

jingber, 2016-08-15:
Audio here is working 100% with the exception of the speaker/headphone muting and the mic (haven't played with these yet).  You have to initially bootstrap the ALSA mixer, but your init scripts should save the mixer states across reboots.  The reason being, those scripts in /usr/lib/alsa aren't perfect for our exact mixer configuration, so pulse is doing the master volume through software mixing. 

Now, are you able to get Kernel Modesetting working on the i915?  This is my **big** problem (no accelerated video).  I also don't any any C/P-state issues, but I suspect these are linked to i915 and why they haven't bit me yet.

The network performance is really, really good on 4.8 now that the sdhci patches are upstream - you just need this one patch and you're golden - https://github.com/hadess/rtl8723bs/files/379444/0001-My-changes-for-4.7.0-rc7-for-Baytrail-T.patch.txt

nicman23, 2016-08-15:
Thanks for all of your work !

Really do not have the time to tinker right now and I do really appreciate your testing and help.

About alsa yes I know that I do not have to rerun the scripts, I was wondering about automatically switching from headsets to speakers.

For modesetting, want problem are you having? You want to use modesetting for X? I think, didn't have any reason to test it, that it used i915. Ttys were working fine and startx didn't read any errors. Does GLAMOR say that is not supported?

About the s states, I meant better suspend support and adding instant2go sleep that is so good on windows.

jingber, 2016-08-15:
Gotcha.  I'm pretty confident I know how to get the rest of ALSA/pulse working (device switching), it's just a little labor-intensive due to the number of mixer controls this thing presents, but I do need the headphone to work so it's up on my priority list.

As for modesetting, I'm getting the classical blank screen when the i915 loads.  This is presumably due to lack of proper MIPI DSI support for this panel (at least in the Flexx 9).  You can load X, but it's on top of any of the unaccelerated framebuffer devices.

S/states, I haven't even tried to suspend the machine.

jingber, 2016-08-15:
It would be really, really helpful, whenever you get a chance (no rush!!), to post your dmesg up to the point that your X server loads, with "options drm debug=12" set.  This way, I can see how the panel *should* be detected and used.  I know how the Asus and the Dell tablets should work, but not these.

nicman23, 2016-08-16:
Yes OK. I might be able today. 

X and kms always worked for my machine, but again it is a different model with mostly the same bios and internals.

Might be distro or .config dependant?

jingber, 2016-08-16:
https://bugs.freedesktop.org/show_bug.cgi?id=82880 ( <- that's exactly what my problem appears to be).  Could also be PEBKAC.

I suspect it's a firmware problem, as i915 is detecting the build-in panel as VGA-1.  I think it should be DSI-1.

burzumishi, 2016-08-17:
You're right "xrandr" shows it up as DSI-1 in my device, and i915_drm module is loaded.

Have you tried to change GRUB_GFX_PAYLOAD=keep and GRUB_GFXMODE="your device resolution" in /etc/default/grub?

I'll post my dmesg while X server is loading with "options drm debug=12" set up.

nicman23, 2016-08-17:
yeah i ll do that today about 10 pm utc+3

burzumishi, 2016-08-17:
I've attached my files to KMS issue: https://github.com/burzumishi/linux-baytrail-flexx10/issues/8

RussianNeuroMancer, 2016-11-18:
There is update at https://bugzilla.kernel.org/show_bug.cgi?id=88471

AXP288 patch published: https://bugzilla.kernel.org/attachment.cgi?id=245031

jharrison022, 2016-11-18:
Outstanding!!  Thank you.

jharrison022, 2016-11-18:
I can confirm this is working.  You have to manually load some of the kernel modules as stated in the last post: https://bugzilla.kernel.org/show_bug.cgi?id=88471

You also need to blacklist the 'battery' kernel module.

Minor fix for 4.9 kernels: http://home.ix.netcom.com/~jjingber/0020-axp288_power_monitoring.patch

nicman23, 2016-11-19:
nice will close when patch is documented.

RussianNeuroMancer, 2016-11-20:
Turns out this patch from my previous comment was insufficient - battery info read once on boot and then doesn't updated anymore. Additional change is required: http://4pda.ru/forum/index.php?showtopic=650808&view=findpost&p=55252765 
Also there is other change needed to hide not-existing battery: https://github.com/sfumato77/Kernel-4.8_Android-x86_BayTrail/commit/e8e70e9722bb7a83d6ee29bc660c393885f47bd1 (three diffs above). Without it two batteries will be detected even if there is really only one.

jharrison022, 2016-11-20:
Thank you.  It is updating here, however.

RussianNeuroMancer, 2016-11-20:
It's updating for you with 4.8 or 4.9?

youling257, 2016-11-21:
i must need to know Charging current, discharge current ， i must use https://github.com/zhaofengli/iwork8/blob/master/scripts/battery.sh

youling257, 2016-11-21:
Since Ubuntu support for reading multiple batteries ，i must use /sys/class/power_supply/test_battery、axp288_fuel_gauge

youling257, 2016-11-21:
modified:drivers/acpi/ac.c，modified:drivers/acpi/battery.c，modified:drivers/acpi/scan.c，only for Android healthd，https://groups.google.com/d/msg/android-x86/9gLesRMykzw/VRs2Lw8yBQAJ

jharrison022, 2016-11-28:
Well, I spoke to soon.

The /sys entries work (/sys/class/power_supply/axp288_fuel_gauge/capacity), but user-space tools don't seem to pick this up.  

Also, user-space won't detect removal of the AC adapter if the axp* kernel modules are loaded when the adapter is present.

At the moment, the axpd script works while this really doesn't.

jharrison022, 2016-11-28:
The /sys entries work so I don't why user space tools are being fussy with this.

youling257, 2016-11-28:
just use https://bugzilla.kernel.org/attachment.cgi?id=245691 !

jharrison022, 2016-11-28:
That patch doesn't really work either --

The meter starts at 95%, and bounces up an down to nonsensical values (ie.  it'll go from 83% to 88% while on battery).

As ugly as it is, the script still works better.

RussianNeuroMancer, 2016-11-28:
Yep, previous one https://github.com/burzumishi/linux-baytrail-flexx10/issues/11#issuecomment-261558368 https://github.com/burzumishi/linux-baytrail-flexx10/issues/11#issuecomment-261760373 works better for me too.

Packages is here: https://yadi.sk/d/pW1Fbl-VyVPoR (4.8.10 rev1).

Tuxman2, 2016-11-30:
Hello Guys,

well, what is it safe or not safe in the patches for Audio, Battery, Backlight and others things at Baytrail hardware ? 
You gave a link for axp288 patch but it can break the hardware (it says USE AT YOUR OWN RISK). I'm ready to test the patches but I don't want to break my hardware.

Thanks.

nicman23, 2016-11-30:
audio and backlight, yes. battery i do not know, but the worst it could 
happen is a kernel panic or that the tablet shuts off because it is out 
of battery and the script did not work :P.

take care that you do follow the audio instructions to the letter (see 
later posts in the respective issue), because if you are stupid (like me 
for example) you may kill your speaker.

Tuxman2, 2016-12-01:
@nicman23: I'm happy that the community found a solution for Baytrail Audio card. I don't know who use an Ubuntu or similar distribution here (I think youling257 and RussianNeuroMancer). Well, it is a pity that the Baytrail hardware doesn't work totally 'out-of-the-box' for now. We must always check the compatibility of the hardware with Linux, while under Windows, it works without problem. Fortunately, the Linux community is still there to solve the problems. Do you put a how-to here when the final kernel 4.9 is released ?

nicman23, 2016-12-01:
> while under Windows, it works without problem

Lol nope

Also yes we will update this for 4.9 and this is not Ubuntu or debian instructions. We (I) try to keep it distribution agnostic.

Haxk20, 2016-12-05:
mine only says battery even the i2c script is not working for me idk if i need anything in kernel config like the AXP288 i think i need this but still those patches and things i hate them because update comes and again you miss new stuff and need to comile again will be happy if ubuntu enable it as default

RussianNeuroMancer, 2016-12-08:
@Tuxman2
Fetch source code: git clone -b 'v4.8.12' --single-branch --depth 1 git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack 
Download relevant Canonical patches for same release: http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.8.12/

Apply patches (run this in mainline-crack folder)
```
for file in /patch/to/mainline-crack/*.patch
do
    patch -p1 < "$file"
done
```

Disable zfs in debian.master/rules.d/amd64.mk
Copy files from debian.master to debian folder. Rename debian/control.stub.in to debian/control.
chmod +x debian/rules
chmod -R 700 debian/scripts
chmod -R 700 debian/scripts/misc
fakeroot debian/rules clean
fakeroot debian/rules editconfigs
Change kernel options (use / for search) or load config file by placing it into build folder.
Pay attention to options I mentioned in Readme: https://yadi.sk/d/pW1Fbl-VyVPoR
Then build kernel and headers packages: fakeroot debian/rules binary-headers binary-generic

Tuxman2, 2016-12-09:
@RussianNeuroMancer: Thanks. ;-)
I'm waiting the final version of 4.9 kernel who will be available in the 2 or 3 days (I hope).
About patches, must I apply all patches (available here and the mainline patches from Ubuntu kernel ppa ) at the same time ?
Is the make oldconfig, make menuconfig and make clean commands are necessary or not after fakeroot debian/rules editconfigs command ?

Regards.

RussianNeuroMancer, 2016-12-09:
@Tuxman2
> must I apply all patches (available here and the mainline patches from Ubuntu kernel ppa ) at the same time ?
I apply patches in order that you can see in Readme.txt

Is the make oldconfig, make menuconfig and make clean commands are necessary or not after fakeroot debian/rules editconfigs command ?
Just  fakeroot debian/rules editconfigs is enough.

Tuxman2, 2016-12-09:
@RussianNeuroMancer: Ok. I will try when the final 4.9 kernel will be available. ;-)

Tuxman2, 2016-12-11:
Hello,

@RussianNeuroMancer: I saw informations about debian rules methods on the net.
 It seems that if we want to use our config file (from the previous kernel), we must copy this file to /debian/config/amd64 or /debian/config/i386/  and then run these commands: 
fakeroot debian/rules clean
debian/rules updateconfigs or fakeroot debian/rules updateconfigs

About kernel build command, we can add skipabi=true to skip ABI checks. I don't know if the AUTOBUILD=1 option is necessary and if we must specify DEB_BUILD_OPTIONS=parallel=x (x is the number of processor/cores the hardware can use for the compilation). This last command can speed up the compilation.
With these options, the kernel build line must be like that:  DEB_BUILD_OPTIONS=parallel=4 AUTOBUILD=1 skipabi=true fakeroot debian/rules binary-headers binary-generic

What do you think about it ?

RussianNeuroMancer, 2016-12-12:
@Tuxman2 
> It seems that if we want to use our config file (from the previous kernel), we must copy this file

To /debian/config/amd64, /debian/config/i386/ or /build, all ways probably will works, but I tested only /build way.

> I don't know if the AUTOBUILD=1 option is necessary
> Is the binary-perarch option must be added ?

Works for me without it.

> if we must specify DEB_BUILD_OPTIONS=parallel=x (x is the number of processor/cores the hardware can use for the compilation)

Nope, building with fakeroot debian/rules binary-headers binary-generic utilize all cores without this variable.

Tuxman2, 2016-12-12:
@RussianNeuroMancer: Ok. ;-) I guess you don't use skipabi=true option anymore ?

@ALL: Good news !!! Final kernel version 4.9 is out (4.9 is now stable). Now we must now what patches can be used or not. :-)

Regards.

RussianNeuroMancer, 2016-12-12:
> I guess you don't use the skipabi=true option anymore ?
Never used it, never heard about it before.

Tuxman2, 2016-12-12:
@RussianNeuroMancer: As the stable 4.9 kernel is out. I guess you will give informations like you did for 4.8 kernel (see your readme.txt file) ?

RussianNeuroMancer, 2016-12-12:
@Tuxman2, probably, but not soon for sure. When - it's depend on all people who write/adjust all this patches.

Tuxman2, 2016-12-13:
Hello,

@RussianNeuroMancer: I forgot to ask you a question the last time. I will use the configuration file of my old kernel (4.4.x) on the new kernel 4.9.x. You told me to put this file (.config I guess) in the /build folder. OK. But as you know, my old configuration file needs to be updated with the latest version of the kernel. With the old method, we put the old kernel configuration file at the root of the kernel source folder, then we use the make oldconfig command, and the make menuconfig command. How does it works with Debian scripts? As you said we must put the old configuration file in /build folder, but then do we use the fakeroot debian/rules updateconfigs and fakeroot debian/rules editconfigs commands? It is this part that I don't quite understand.

Thank you for the clarification and your help. ;-)

Tuxman2, 2016-12-18:
Hi,

I see that 4.8.0-30.32 kernel is available in the Ubuntu repositories. 4.8.0-30.32 kernel sources are also available. If I want to use that kernel version or create a custom kernel from that version, do I need to apply your patches on 4.8.0-30.32 kernel sources (clocksource patch for audio, axp288 patches for battery, patches for backlight, ...) ? Is it easy to compile kernel source file from Ubuntu repositories ?

Thanks.

RussianNeuroMancer, 2016-12-20:
@Tuxman2 

> my old configuration file needs to be updated with the latest version of the kernel

Do this manually in fakeroot debian/rules editconfigs or save config, edit it, load edited config.

Tuxman2, 2016-12-22:
@RussianNeuroMancer: ok, but I would have preferred to do it automatically.
Could you provide all the necessary patches files in this link: https://yadi.sk/d/pW1Fbl-VyVPoR ?
You gave the links in your readme.txt but it is not easy to download them.

Thanks in advance.

jharrison022, 2016-12-23:
RussianNeuroMancer patches are working for me (I tested for awhile before I opened my mouth this time).

Thank you.

jharrison022, 2016-12-23:
This may make it easier than going through this whole thread.  Tested on 4.9:

http://home.ix.netcom.com/~jjingber/0020-axp288_power_monitoring.patch
http://home.ix.netcom.com/~jjingber/0022-hide_phantom_batteries.patch

youling257, 2016-12-23:
can't open it http://home.ix.netcom.com/~jjingber

jharrison022, 2016-12-23:
<i>can't open it http://home.ix.netcom.com/~jjingber</i>

Sorry -- should work now!!  Thank you.

jharrison022, 2016-12-23:
youling ^-- actually these seem to be based on your scripts.  They weren't working on 4.8 but that are working on 4.9.

Tuxman2, 2016-12-23:
I think I will use the official kernel sources from Ubuntu (for now, it is 4.8.0-34) and apply your custom patches on it. I will preferred waiting for the 4.9 kernel later. Thanks to RussianNeuroMancer to give the patches links in his own readme.txt file but it doesn't help me because it is not .patch files.
Can you tell me what patches must I apply if I want to use the kernel sources 4.8.0-34 from Ubuntu ?

Thanks.

jharrison022, 2016-12-25:
The clock source patch is not a requirement.  If you enable MCLK_EN, then you need the patch.

nicman23, 2016-12-25:
Can you explain what the MCLK_EN is?

jharrison022, 2016-12-25:
That's just the constant that enables the audio driver to use the Baytrail platform clock patch.  It's unnecessary and confusing.

Tuxman2, 2016-12-27:
It will be great if someone here could make a summary on what to apply as modifications/patches for 4.8 kernel and 4.9 kernel (to make a custom kernel) in order to operate our devices.  There is so much information in the different messages that we are lost.

Thanks.

nicman23, 2016-12-28:
http://mailman.alsa-project.org/pipermail/alsa-devel/2016-December/115886.html

relevant

[27 more replies]

closed 2017-03-17 by nicman23