burzumishi/linux-baytrail-flexx10 #12: Documentation Updates

nicman23, 2016-08-18:
Made a branch for updating the readme

Atm I have mostly removed any obsolete information (ie c states problem) and currently waiting for 4.8 to proceed.

burzumishi, 2016-08-18:
OK, thanks!

I have been updating the readme a bit too, I'm moving things from the main readme to the Wiki pages.

When I get home I'll test DKMS for wifi.

nicman23, 2017-03-17:
- [ ] Battery from #11 
- [ ] Audio from #7 
- [ ] Wifi / clocksource / etc from #33 

Wishlist:

- [ ] write deb and pkgbuilds for firmware / wifi / kernel

nicman23, 2017-03-18:
@Tuxman2  @RussianNeuroMancer @jusss please move all discussion here for bugs that are already fixed.

please tag anyone that i missed from this repo.

Mainline 4.10rc1 with firmware from git has out of the box audio and battery (you need to enable it with a config).

I am now looking into the wifi. I ll make a slim .config with only the modules that our machines need and i ll upload it in the repo (i ll write here to inform you)

Haxk20, 2017-03-18:
I do you need some mmc patch and then it didnt freeze but i lost it because of update and reinstall but will try to find it :)

RussianNeuroMancer, 2017-03-19:
@nicman23
deb packages for WiFi and Bluetooth is there: 

https://launchpad.net/~russianneuromancer/+archive/ubuntu/drivers
https://yadi.sk/d/LbaBpYVuqYfCy

nicman23, 2017-03-19:
btw upstream linux-firmware has the needed firmware files for 8723bs

Haxk20, 2017-03-19:
Reinstalled arch will build kernel in few days but noticed that 4.10.3 kernel without kernel parm didn't froze and even tsc is selected as good clocksource and yeah its working nicely will see what 4.11 rc2 with added axp288 configs will do if it will work then I will probably have working WiFi (never warning here perfectly)

RussianNeuroMancer, 2017-03-20:
@Haxk20 

> will try to find it :)

You mean [this one](https://github.com/hadess/rtl8723bs/issues/76#issuecomment-272932161)?

nicman23, 2017-03-21:
even with that patch, the 8723bs driver cause the tablet to freeze :/

I have been using it regularly the couple past days and had no issues without the wifi driver ( i just use and usb wifi nic )

also, again, for those who have wpa_supplicant > 2.5 , you will have issues with NetworkManager.

jusss, 2017-03-21:
@nicman23 @RussianNeuroMancer @Haxk20  I wonder did 8723bs driver from hades cause random freeze? I used linuxium's ISO image and wifi work normal, no random freeze

youling257, 2017-03-21:
i used ax88179_178a ，had never freeze，8723bs only 70MBps，i use ax88179_178a 150Mbps

beevabeeva, 2017-03-21:
I've been using Linuxium's older image ( 4.9 I think.) I don't have sound (despite his instructions on installing patch) - do you have sound?
He hasn't made an image for the latest kernel last I checked.
I wish he'd make a tutorial explaining how he makes the image in the first place.
If someone knows of one, a link would be appreciated.

RussianNeuroMancer, 2017-03-21:
@nicman23 

> even with that patch, the 8723bs driver cause the tablet to freeze :/

With 4.11?

jusss, 2017-03-22:
@beevabeeva  http://linuxiumcomau.blogspot.jp/2017/01/bleeding-edge-ubuntu-4100-57-kernel.html?m=1     I use his ubuntu 17.04 lubuntu,  and after I remove pulseaudio and config ~/.asoundrc, speaker will work, but headphone no, wifi ok, no random freeze

RussianNeuroMancer, 2017-03-22:
Linux 4.11rc3 build with patches: https://yadi.sk/d/pW1Fbl-VyVPoR
Please report freezes to https://bugzilla.kernel.org
Please report GPU hangs to Intel: https://01.org/linuxgraphics/documentation/how-report-bugs
Thanks.

jusss, 2017-03-22:
> documentation/how-report-bugs

thanks!

Haxk20, 2017-03-22:
I contacted linuxium at email if he can give us only patches and he only talked about donate and if I ask second time for 4.10 he didn't respond.
If you contact him on email ansd he will send you only patches and not link to patched kernel send it here please

jusss, 2017-03-22:
@Haxk20  reference this article   http://linuxiumcomau.blogspot.jp/2017/01/bleeding-edge-ubuntu-4100-57-kernel.html?m=1    maybe we can find these patches from github or bugzilla.kernel.org

RussianNeuroMancer, 2017-03-22:
Check Readme.txt in Linux 4.11rc3 folder above.

beevabeeva, 2017-03-22:
Yea, I don't think he writes the patches himself (does he?). If he does, then I wouldn't mind buying him a beer in exchange for them... But more valuable would be learning how to go about creating patches/ drivers in C...

Tuxman2, 2017-03-22:
In my case I fixed the sound and microphone problem. I must use IN3_MAP and ssp0-aif1. 
I will send the quirk informations to plbossart so that he integrates them to the 4.11-rcx kernel.

Thanks to all people here for the informations above. ;-)

RussianNeuroMancer, 2017-03-22:
@Tuxman2 my bad, I'll upload it tomorrow.

Tuxman2, 2017-03-22:
@RussianNeuroMancer: OK, thanks. ;-)

@All: I'm currently doing a how-to for those who want to create a custom kernel for their hardware (Baytrail tablet in our case) with Ubuntu and derived distributions. I hope to give you this how-to as soon as possible. ;-)

beevabeeva, 2017-03-22:
@Tuxman2 Ah great! Thank you very much and looking forward to learning from your tut!

nicman23, 2017-03-22:
@RussianNeuroMancer 
> With 4.11?
yes

> Bugreport filled anywhere?

also yes see https://github.com/hadess/rtl8723bs/issues/113

RussianNeuroMancer, 2017-03-22:
@nicman23 
Thanks!
So two workarounds is necessary:
- revert [one commit](https://w1.fi/cgit/hostap/commit/?id=14220fe64c4e8180380d9fd15a985aa699b00e2f). Probably if driver doesn't get fixed I will provide package with workaround for Ubuntu 17.10 (shouldn't be needed for 17.04, as I understand). 
- [edit Network Manager config](https://github.com/hadess/rtl8723bs/issues/113#issuecomment-284191105).

RussianNeuroMancer, 2017-03-23:
@Tuxman2 files re-uploaded.

@nicman23 Is there bugreport about freeze with WiFi driver on kernel with sdhci patch?

Tuxman2, 2017-03-23:
@RussianNeuroMancer: Ok, thanks. ;-)

@All: Ok, here is the little tutorial I made for people who wants to have a custom kernel for their Baytrail machine. This tutorial is mainly for Ubuntu and derived distributions (ex: Linux Mint).

[HowToCustomKernelUbuntu.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/864787/HowToCustomKernelUbuntu.txt)

Good compilation. ;-)

RussianNeuroMancer, 2017-03-23:
@Tuxman2 on some tablets and laptops there is blank screen as soon as Xorg started. Issue is present up to Linux 4.11. Intel fixed that, but fix in drm-intel-next (so it well get into 4.12, two months laters). rev2 for such devices and for people who doesn't want to wait.

Tuxman2, 2017-03-23:
@RussianNeuroMancer: Ok.

@All: Is someone here can tell me what this dmesg error means : 'i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed', it would be nice. :-) 
I don't find any informations about it on the net. :-(

nicman23, 2017-03-23:
@RussianNeuroMancer  yeah i know about the network manager config, but thanks about the flush hostap patch :)

@All btw i have been using this little tablet for the patch week or show as my main uni machine (without the wifi driver, with an external usb wifi interface) and the experience is quite pleasant. as of now i have not had a single crash

beevabeeva, 2017-03-23:
@Tuxman2 Thank you so much!

RussianNeuroMancer, 2017-03-24:
@nicman23 since I can not reproduce WiFi freeze with patched kernel, could you please fill bugreport about this? (To hadess repo, I guess?)

jusss, 2017-03-24:
where I can get "some SDIO/I2C related patches from Hans De Goede." to anti-freeze?

jusss, 2017-03-24:
@RussianNeuroMancer I saw there are linux-image-generic.deb and linux-image-extra.deb, do I need install them all? also linux-headers-extra,  and so far I haven't meet a random freeze with you kernel,  :-)

RussianNeuroMancer, 2017-03-24:
@jusss 
> do I need install them all?

AFAIK yes.

> also linux-headers-extra, and so far I haven't meet a random freeze with you kernel, :-)

Please install Opera, open opera://flags, enable "Override software rendering list", set "GPU rasterization" to "Force-enabled for all layers", set "Number of raster threads" to 4, restart Opera, then surf web, watch online video, etc. I get freeze, and it would be very helpful if anyone else besides me will able to reproduce this issue.

nicman23, 2017-03-24:
@RussianNeuroMancer same settings on chromium has been stable for me (opera is basically old chromium)

RussianNeuroMancer, 2017-03-24:
Very strange, freezing for me on two devices. Even Spotify client with Chromium inside cause freeze.

nicman23, 2017-03-24:
I run spotify for hours playing things. I must add again that i do not have the internal running wifi but a usb device.

jusss, 2017-03-25:
@ALL  dmesg show lots same messages like "IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready     RTL8723BS: nolinked power save enter/leave",  driver from hades, kernel from RussianNeuroMancer, have I miss something?

jusss, 2017-03-25:
same with linuxium's ISO

RussianNeuroMancer, 2017-03-25:
@jusss could you please look into [this](https://github.com/burzumishi/linux-baytrail-flexx10/issues/12#issuecomment-289006961) message?

For WiFi issue try [this](http://www.studioteabag.com/science/dell-venue-pro-linux/) (Ctfl+F, "SDHCI ACPI").

jusss, 2017-03-25:
@RussianNeuroMancer I don't have my tablet now, I will give you the test result tomorrow, also wifi messages,  :-)

jusss, 2017-03-26:
@RussianNeuroMancer  I test opera as you said and no freeze on my tablet, 
the wifi issue, I set
/sys/bus/platform/drivers/sdhci-acpi/80860F14\:00/power/control 
on
/sys/bus/platform/drivers/sdhci-acpi/80860F14\:02/power/control 
on
/sys/bus/platform/drivers/sdhci-acpi/INT33BB\:00/power/control 
on
and it still pops lots message up on dmesg until I connect it to wifi,
I mean if I don't connect it to wifi, it pops up, I connect it to wifi then it won't pops up

Tuxman2, 2017-03-26:
@RussianNeuroMancer: About the quirk for the sound problem, I've got a patch. It seems that with this patch you don't need to add the quirk for your sound hardware. I need to test it. If it works I will give you the patch.

jusss, 2017-03-27:
@ALL finally I meet a freeze with RussianNeuroMancer's 4.11-rc2 kernel, I was opening chromium and emacs, and using sudo vim /etc/udev/rules.d/80-udisks.rules on ansi-term mode of emacs, suddently it freeze...   :-(  
I think it is wifi driver's issue, if I'm wrong please tell me

jusss, 2017-03-27:
@RussianNeuroMancer @ALL
lots message pop on dmesg like 
axp288_fuel_gauge: Error reading 0xbc :axp288 reg read err:-110  
i2c_designware 80860F41:04 punit semaphore timed out, resetting, PUNIT SEM:2, couldn't acquire bus owership
and I run top
root PR -51 CPU 25.7%   command irq/46-mmc0
the kernel is RussianNeuroMancer's 4.11-rc2, anyone meet this?

Tuxman2, 2017-03-27:
@jusss: Try to compile the 4.11-rc4 kernel version with the patches and the options given by RussianNeuroMancer.

RussianNeuroMancer, 2017-03-27:
I guess patches from Hans tree are needed.

There is 4.11rc2: https://github.com/jwrdegoede/linux-sunxi/tree/bq24190-wip-not-rebased

youling257, 2017-03-27:
platform: x86: silead_dmi: Add entry for Insyde 7W tablets
@RussianNeuroMancer  ，my tablet silead always error -11，https://bugzilla.kernel.org/show_bug.cgi?id=193721 ，my tablet silead no "Shutdown GPIO request " 
https://patchwork.kernel.org/patch/9531463/ ，“On some x86 tablets we cannot directly access” ，isn't my tablet .

youling257, 2017-03-27:
Jose Luis s  has 3 chuwi tablets all dual android 4.4.4 (android-ia), windows  8.1/10
vx8 3g business (2GB ram 32 GB emmc, GPS, 3g)
v89 (2GB ram 64 GB emmc, GPS, 3g)
Hi8

发件人：	Jose Luis <joselusar@gmail.com>
收件人：	youling 257 <youling257@gmail.com>
日期：	2017年3月16日 下午9:37
主题：	Re: Mensaje privado sobre: Porting CyanogenMod 12.1 to x86

“Hi Youling,  First I need to test my silead firmware files with a working kernel and a working iso, so I pick a cm13 x86-64 iso with 4.8 kernel from Henri (closest to vor kernel version 4.8.17), and begin to test. Using silead.ko I got a "shutdown GPIO request failed" error -16 .”

"shutdown GPIO request failed" happen on chuwi tablet .

RussianNeuroMancer, 2017-03-27:
@youling257 Silead driver works for me with this kernel: https://yadi.sk/d/pW1Fbl-VyVPoR (4.11rc3).

youling257, 2017-03-27:
@RussianNeuroMancer ，silead work well for me with 4.8.17 Android x86 kernel ，i am using Android x86 lineage os 14.1

youling257, 2017-03-27:
@RussianNeuroMancer ，https://github.com/jwrdegoede/linux-sunxi/commit/be11c66d43299206aa0f8056fab0d25acdd99798 ，how to “invert_x=yes  invert_y=yes   swap_x_y=yes” ？

Haxk20, 2017-03-28:
Hi guys do anybody here have a problem like me that when im watching some YT video and right part of screen maybe 5cm from the edge it froze and i need to move mouse there to unfreeze it and its very anoying can you guys help me and the patch in kernel/patches/v4.8/0001-mmc_sdhci_on_intel_baytrail.patch help me to unfreeze the device when wifi is on 2 days and no freeze

jusss, 2017-03-28:
do you rotate your screen? tried another DE or wm?

nicman23, 2017-03-28:
@Haxk20 probably a compositor issue, try a different DE

Haxk20, 2017-03-28:
Fixed by using chromium probably Firefox issue and another problem WiFi is not finding any AP I will post dmesg later today but I looked into it and something about IPv6 and wlan0 is down applyed patches but 1 of them failed and still cannort find AP

RussianNeuroMancer, 2017-03-28:
Patches from Hans for 4.11: https://yadi.sk/d/pW1Fbl-VyVPoR (zip in 4.11 folders). 
Source: https://github.com/jwrdegoede/linux-sunxi/commits/bq24190-wip-not-rebased

Haxk20, 2017-03-28:
Thanks not used them and now WiFi is working and watching yt without problem by disabling IPv6 changing in network manager.conf and applying patches in rtl8723bs folder
Well I have functional tablet almost everything works

[458 more replies]