burzumishi/linux-baytrail-flexx10 #14: Right Speaker started to distort sound!

burzumishi, 2016-08-18:
My right speaker has started to distort all sounds while playing video, it seems to be broken. Rebooting on Windows did not solve the problem, the distortion was there.

Rebooting back to Linux and reviewing alsamixer settings I was able solve the issue, I found a MIX channel for Right Speaker active, disabling it all went back to normality.

I'm wondering how this setting could change while playing a video ...

jingber, 2016-08-19:
I had this exact issue last night, after I updated my Debian Stretch install, which included an updated Pulse Audio.  

The sound was distorted until I re-installed the original mixer settings into /var/lib/alsa and ran alsactl restore.

burzumishi, 2016-08-19:
I think something is wrong with the gain settings of output channels, it only happens with the right speaker ... my sound is distorted again -.-

Well at least, I guess the speaker is not broken, I'll restore mixer settings later (now I'm at work).

closed 2016-12-24 by nicman23