burzumishi/linux-baytrail-flexx10 #15: Screen brightness

youling257, 2016-09-04:
really can work ?

nicman23, 2016-09-04:
have to tested, did not even install a gui last time i run linux.. Worse case scenario you adjust it with xset.

jharrison022, 2016-09-06:
The i915 driver needs a patch to make this work.  I have it and I'll post it up next time I crack open the tablet.

youling257, 2016-09-07:
please post patch

jharrison022, 2016-09-09:
http://home.ix.netcom.com/~jjingber/0007-intel_baytrail_backlight.patch 

i915 kernel module needs i915.force_backlight_pmic=1

youling257, 2016-09-09:
i915.force_backlight_pmic=1，can't adjust screen brightness，my tablet z3735f .
other z3735f test it ?
only use this patch ？.config need xxxx=y or m ?

parkerlreed, 2016-09-20:
@jharrison022 I just wanted to say thanks for the patch. I just used this on 4.8 rc7 from git and it works beautiful on my Asus T100TA.

nicman23, 2016-09-21:
can anyone with a flexx10 confirm that it works and with which kernel revision?

youling257, 2016-09-21:
https://bugs.freedesktop.org/show_bug.cgi?id=85977
https://bugs.freedesktop.org/show_bug.cgi?id=96571
https://bugs.freedesktop.org/show_bug.cgi?id=90075
pwm backlight

i don't know their discussion results

jharrison022, 2016-09-23:
I think it's safe to say the screen on the Flexx 9 and 10-style tablets are very different.  This patch works over here.

jharrison022, 2016-09-23:
I think that's really the only difference in these tables, expect for the Flexx 9's sh*tty firmware and SSD size.

youling257, 2016-09-23:
I'm not good at English，I can't understand their discussion for pwm backlight，
the problem is [drm:pwm_setup_backlight [i915]] *ERROR* Failed to own the pwm chi

nicman23, 2016-09-24:
@jharison22 so it works with 10 and not with 9

jharrison022, 2016-09-25:
It works on the 9, at least.  No idea about the 10.

RussianNeuroMancer, 2016-11-10:
Try this patches: https://www.tinc-vpn.org/pipermail/click/2016-October/000347.html

Build options: 

```
CONFIG_PWM=y
CONFIG_PWM_CRC=y
CONFIG_I2C_DESIGNWARE_PLATFORM=y
CONFIG_I2C_DESIGNWARE_PCI=y
CONFIG_INTEL_SOC_PMIC=y
CONFIG_DRM_I915=m
CONFIG_PWM_LPSS=y
CONFIG_PWM_LPSS_PCI=y
CONFIG_PWM_LPSS_PLATFORM=y
CONFIG_X86_INTEL_LPSS=y
```

youling257, 2016-11-10:
https://bugs.freedesktop.org/show_bug.cgi?id=96571 ，do you read this bug ? @RussianNeuroMancer

RussianNeuroMancer, 2016-11-10:
Yes. @youling257, do you refer to some particular comment?

youling257, 2016-11-10:
https://www.tinc-vpn.org/pipermail/click/2016-October/000347.html  ，he edit DSDT，i need know DSDT patch .

DSDT patch，@zefie ，half year ago，get backlight work，https://files.persona.cc/linux/intel_baytrail_soc/nextbook%2010.1/android/android_x86_010c.txt

youling257, 2016-11-10:
@RussianNeuroMancer  ，https://github.com/zefie/nxt_and_x86/blob/x86/files/generic/boot/boot/dsdt/dsdt_changes.patch

DSDT patch，@@ -433,9 +433,10 @@ ，@@ -13764,10 +13765,10 @@，@@ -13731,10 +13732,10 @@，can make Volume Buttons work ；don't use DSDT patch，use this patch，https://patchwork.kernel.org/patch/6118791 ，Volume Buttons also work .
two method make Volume Buttons work .

so，if @@ -9599,31 +9600,7 @@ and @@ -9656,32 +9633,7 @@ is for backlight，but my dsdt can't patch @@ -9599,31 +9600,7 @@ ，@@ -9656,32 +9633,7 @@ ，get dsdt.dsl.rej file .

RussianNeuroMancer, 2016-11-10:
@youling257 

> he edit DSDT

Yep, but as I understand most BYT devices doesn't need DSDT changes to get backlight work. This is wrong assumption? 

> don't use DSDT patch，use this patch，https://patchwork.kernel.org/patch/6118791

This patch is old, why it's not merged into upstream? Is there bugreport about this?

> two method make Volume Buttons work 

One method is this patch https://patchwork.kernel.org/patch/6118791 and another? DSDT edit or something different?

youling257, 2016-11-10:
@RussianNeuroMancer  , most BYT devices  , what byt ? z3735f [drm:pwm_setup_backlight [i915]] *ERROR* Failed to own the pwm chip .

RussianNeuroMancer, 2016-11-10:
@youling257

> what BYT devices

BayTrail tablets/laptops/transformers

> ERROR Failed to own the pwm chip .

Check this: https://drive.google.com/drive/folders/0B_uxdqSfzNGwOG1UUEh2ckM4LVU

youling257, 2016-11-10:
@RussianNeuroMancer , i don't will go to test his kernel and modules , i must solved z3735f android x86 backlight problem .

youling257, 2016-11-10:
https://drive.google.com/drive/folders/0B_uxdqSfzNGwTjF2M2w2VDBzb0k?usp=sharing ，“BUT for make this set of patches work, I had to modify my DSDT table” @RussianNeuroMancer

RussianNeuroMancer, 2016-11-10:
I not sure, but maybe DSDT modification could be necessary for his device but not necessary for some other devices? Did you tested it?

RussianNeuroMancer, 2016-11-10:
@youling257

> i don't will go to test his kernel and modules

Okay, maybe someone else will test this approach.

youling257, 2016-11-10:
@RussianNeuroMancer  ，please you tell me，i test his deb file，has what use ? 
i need more patch，his https://drive.google.com/drive/folders/0B_uxdqSfzNGwTjF2M2w2VDBzb0k?usp=sharing ，no use，i recompile my kernel and modules use these three patch，backlight doesn't work .

youling257, 2016-11-10:
thanks for zefie DSDT patch，https://github.com/zefie/nxt_andx86_kernel/commit/121736b5fc95c2325c13ff5c75a3cd980192ef56 https://github.com/zefie/nxt_and_x86/blob/x86/files/generic/boot/boot/dsdt/dsdt_changes.patch
Android x86 backlight can work，brightness control can work .

RussianNeuroMancer, 2016-11-10:
@youling257

Just to confirm:

1. With this three patches https://drive.google.com/drive/folders/0B_uxdqSfzNGwTjF2M2w2VDBzb0k backlight control doesn't work for you,

2. But with this three patches and also DSDT patch, backlight control DO WORK for you? Is that correct?

What your device? Nextbook Flexx 10.1"?

youling257, 2016-11-11:
let me tell you，there is z3735f problem，i only care z3735f problem . @RussianNeuroMancer

RussianNeuroMancer, 2016-11-11:
@youling257

> i only care z3735f on Android x86 problem

As I understand you solved it, right? I want to clearly understand the way you do this. This is why I asking you this questions from my previous comment. This information could be very helpful for other people.

youling257, 2016-11-11:
“@RussianNeuroMancer ，zefie/nxt_andx86_kernel@121736b ，https://github.com/zefie/nxt_and_x86/blob/x86/files/generic/boot/boot/dsdt/dsdt_changes.patch

DSDT patch，@@ -433,9 +433,10 @@ ，@@ -13764,10 +13765,10 @@，@@ -13731,10 +13732,10 @@，can make Volume Buttons work ；don't use DSDT patch，use this patch，https://patchwork.kernel.org/patch/6118791 ，Volume Buttons also work .
two method make Volume Buttons work .

so，if @@ -9599,31 +9600,7 @@ and @@ -9656,32 +9633,7 @@ is for backlight，but my dsdt can't patch @@ -9599,31 +9600,7 @@ ，@@ -9656,32 +9633,7 @@ ，get dsdt.dsl.rej file .”

yes，@@ -9599,31 +9600,7 @@ and @@ -9656,32 +9633,7 @@ is for backlight ，i patch my dsdt.dsl success ，compile dsdt.hex into kernel ，get backlight work .

youling257, 2016-11-11:
https://plus.google.com/116937366969544577777/posts/fc7y2oAirh5

nicman23, 2016-11-16:
please, leave bugs open until proper documentation is provided.

Boteium, 2016-12-15:
I tested the patch for kernel 4.8.3 with modified DSDT compiled in on Teclast X80h. The error message (failed to own pwm chip) disappeared and "xrandr --output DSI-1 --off" did turn off/on the backlight despite putting X into black screen. Backlight control still doesn't work. The only thing works is turninig backlight on and off. However, I still get black screen everywhere. I can bypass this black screen issue with i915.fastboot=1 parameter. But the screen turns black when I "restart X/xrandr rotate/xrandr  on/off". So, my screen only works at first login(logout also causes black screen) and I cannot rotate it.

youling257, 2016-12-15:
CONFIG_PWM=y
CONFIG_PWM_CRC=y
CONFIG_I2C_DESIGNWARE_PLATFORM=y
CONFIG_I2C_DESIGNWARE_PCI=y
CONFIG_INTEL_SOC_PMIC=y
CONFIG_DRM_I915=m
CONFIG_PWM_LPSS=y
CONFIG_PWM_LPSS_PCI=y
CONFIG_PWM_LPSS_PLATFORM=y
CONFIG_X86_INTEL_LPSS=y

youling257, 2016-12-15:
DSDT patch，such as http://pastebin.com/ZKfJNySj
detail https://plus.google.com/116937366969544577777/posts/fc7y2oAirh5 

i compile include/dsdt.hex into kernel，
# CONFIG_STANDALONE is not set
CONFIG_ACPI_CUSTOM_DSDT_FILE="dsdt.hex"
CONFIG_ACPI_CUSTOM_DSDT=y

Boteium, 2016-12-15:
yes, these are exactly what I did. I do have these kernel config parameter the same as above and compiled in the dsdt.hex.

youling257, 2016-12-15:
only use these three patch，https://drive.google.com/drive/folders/0B_uxdqSfzNGwTjF2M2w2VDBzb0k
should be drm:pwm_setup_backlight [i915]] *ERROR* Failed to own the pwm chip: pwm_lpss .
if edit dsdt，should get backlight work .

Boteium, 2016-12-18:
ok, with more testing, backlight is actually working despite showing a black screen. Booting with i915.fastboot=1 gets me a functioning X (without backlight support). Booting without that parameter gets me a black screen with working backlight support.

Tuxman2, 2016-12-24:
Hi,

I can't adjust brightness during boot sequence but I can adjust brightness when the desktop appears (with FN + fonction button for brightness). Then I think I don't need the patch ? Right ?
To adjust the brightness during boot sequence I use a script that I applied with initramfs.

nicman23, 2016-12-24:
do not even have a keyboard for my device......

Tuxman2, 2016-12-24:
If you want the how-to to adjust brightness at boot (with the help of initramfs), tell me I will post here. ;-)

youling257, 2016-12-24:
if you say script ，i don't know it .
the adjust brightness by pwm control from DSDT .
LPSS based device or CRC PMIC

Tuxman2, 2016-12-25:
I don't use xrandr script but another tip. The tip I use is to adjust the brightness after boot. Here is the tip:

1. Create a script file named local-backlight-brightness in /etc/initramfs-tools/scripts/init-premount/
2. Add this content in it:
`#!/bin/sh
PREREQ=""
prereqs()
{
echo "$PREREQ"
}

case $1 in
prereqs)
prereqs
exit 0
;;
esac

# Start of treatment to adjust brightness for Intel card

echo x > /sys/class/backlight/intel_backlight/brightness
exit 0`

Note: replace x with the value you need.
You can check your current value with that command: cat /sys/class/backlight/intel_backlight/actual_brightness
Modify your brightness with the brightness control bar or with FN+fonction button. Recheck the value with the command above. Note that one who will be defined in the code above (echo yourvalue > /sys/class/backlight/intel_backlight/brightness).

3. The script must be executable, and you must update initramfs (in root mode):

# chmod a+rx /etc/initramfs-tools/scripts/init-premount/local-backlight-brightness
# update-initramfs -u -k all (for all the kernels available).

4. Reboot your hardware.

Note: Here is the source I used: http://unix.stackexchange.com/questions/117101/how-to-set-laptop-screen-brightness-just-after-boot-with-full-disk-encryption

youling257, 2016-12-25:
If your dmesg no this , pwm_setup_backlight [i915]] ERROR Failed , the correct method to solved it

youling257, 2017-01-02:
4.10 kernel，if patch intel_panel.c and intel_drv.h ，can't compile ，http://pastebin.com/AwDqbfZC ；4.8 and 4.9 ，no problem，backlight control can work .

4.9 kernel and 4.10 kernel ，

> root@x86:/ # dmesg | grep silead
[   14.031463] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[   14.075816] silead_ts i2c-MSSL1680:00: Registers clear error -11
[   14.087734] silead_ts: probe of i2c-MSSL1680:00 failed with error -11
[  180.627479] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[  180.628877] silead_ts i2c-MSSL1680:00: Registers clear error -11
[  180.638825] silead_ts: probe of i2c-MSSL1680:00 failed with error -11
root@x86:/ #

> [   13.806770] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 68, flags 0x0
[   13.914450] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[   13.920399] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[   13.947868] gslx680: probe of i2c-MSSL1680:00 failed with error -11

shit，I don't have to follow up ，the 4.8 kernel is perfect and exclusively .

Haxk20, 2017-01-15:
What patches are required for backlight to work ?

closed 2017-02-24 by youling257