burzumishi/linux-baytrail-flexx10 #19: OV2680 built-in camera driver

RussianNeuroMancer, 2016-11-10:
Source code of OV2680 driver is available here: https://android.googlesource.com/kernel/mediatek/+/android-6.0.1_r0.110/arch/arm/mach-mt6582/sprout/imgsensor/

zefie, 2016-11-10:
That source tree is 3.x, and I've already tried (and failed) to port the 3.x drivers to 4.4.

nicman23, 2016-11-16:
thanks! Does it work with mainline (or at least a new android-x86 kernel) ?

RussianNeuroMancer, 2016-11-16:
AFAIK it doesn't. Some modifications probably required to get it working on current kernels. 

And, by the way, it was available long time ago in gmin-kernel, like many other drivers for BayTrail devices: https://github.com/android-ia/device_intel_gmin-kernel/raw/master/x86_64/src.tgz

nicman23, 2016-11-16:
oh yes i was aware about that kernel, but had forgotten about it.

sadly, i m no kernel dev to patch them for any kernel.

RussianNeuroMancer, 2016-11-18:
GM0310 driver is also available in same patchset.

jharrison022, 2016-11-18:
Which is the front camera and which is the rear?

RussianNeuroMancer, 2016-11-18:
As I remember OV2680 is read and GM0310 is front.

Tuxman2, 2016-11-18:
Hello, 

I've got the hm2056 and ov5648 hardware.

RussianNeuroMancer, 2016-11-18:
@Tuxman2 
Your transformer is Nextbook Flexx 10.1"?

OV5648 driver is also here: https://github.com/01org/ProductionKernelQuilts/tree/master/uefi/cht-m1stable/patches

Tuxman2, 2016-11-18:
@RussianNeuroMancer : No, it is a Medion Akoya E2212T model.

RussianNeuroMancer, 2016-11-18:
As I see all drivers is available, so it's not secret how to interact with this cameras. But nobody ported this drivers to upstream kernel.

Tuxman2, 2016-11-18:
if I apply the patches for OV5648  (available in the link you give me) I can make my camera work ?

RussianNeuroMancer, 2016-11-18:
Most likely no. At first someone need to modify this drivers to make it work with more newer kernel.

Tuxman2, 2016-11-20:
Here is what I found for ov5648 : https://github.com/NoelMacwan/Kernel-C6806-KOT49H.S2.2052/blob/master/drivers/media/platform/msm/camera_v2/sensor/ov5648.c and https://github.com/NoelMacwan/Kernel-C6806-KOT49H.S2.2052/tree/master/drivers/media/platform/msm/camera_v2/sensor

I don't know if we can use it with Linux kernel.

Tuxman2, 2017-01-12:
No news about the support of OV and HM cameras on Linux ?

nicman23, 2017-01-12:
I will likely never be. There are too small of a number of devices and too low  of quality for anyone to care about a 1MP camera

Tuxman2, 2017-01-15:
@nicman23: It is not a good news. I don't care for the front camera but I would like to use the rear camera.

dfiloni, 2017-01-24:
I have a Lenovo Miix 310, it has a ov2680 front camera and a ov5648 rear camera. I tried to port ov5648 driver from https://android.googlesource.com/kernel/bcm/+/android-bcm-tetra-3.10-lollipop-wear-release/drivers/media/i2c/soc_camera/ov5648.c to mainline however it isn't working (webcam is not recognized at all)
I think it doesn't work due to: https://bugzilla.kernel.org/show_bug.cgi?id=110151 , so I don't think the camera driver is enough to get it working.

Tuxman2, 2017-02-05:
Why is it to difficult for kernel developpers to give support for ov2680, ov5648 and hm2056 cameras ?
I see support for some omnivision cameras in the kernel.

nicman23, 2017-02-06:
Because it is an open source project and no one cares about these cameras as I said :P

Tuxman2, 2017-03-19:
Well, I've got the source code for the HM2056 driver and the OV5648 driver. I made a makefile for each. You can try to compile them (kernel-headers, build-essential and make command are necessary) and test them.

HM2056 driver: https://snf-738721.vm.okeanos.grnet.gr/index.php/s/UtIkL9IJyxGPOXP
OV5648 driver: https://snf-738721.vm.okeanos.grnet.gr/index.php/s/rvBIb7nnfrnzewz

For example, go to HM2056 directory and launch the make command. You have must have a .ko file after compiling. Then, try this:

- Put the hm2056.ko file in /lib/modules/numberofyourkernel/kernel/drivers/platform/x86
- sudo depmod -a
- In /etc/modules file, put the word: hm2056
- Reboot to see if the module is loaded (you can see with the lsmod command).

RussianNeuroMancer, 2017-03-19:
Cameras won't work until Intel Atom Image Signal Process driver get ported to current kernel.

sfumato77, 2017-03-19:
Intel Atom Image Signal Process It has started to be implemented in the kernel  4.11.

look here: https://kernel.googlesource.com/pub/scm/linux/kernel/git/gregkh/staging/+/staging-testing/drivers/staging/media/atomisp/

but there is a problem the two sensors (OV2680 and HM2056) are not yet supported
read TODO files

who owns a compatible CCD sensor can try to compile a new kernel

and let us know. 
unfortunately, I can fill out a atomisp.ko.
but I do not dirvers of HM2056.c sensor OV5648.c me error during build

P.S.

for those who want to try to fill atomisp for kenrel <411
You can give it a try as well:

copy dir atomisp
in /drivers/media/staging

Add in Makefile
obj-$(CONFIG_INTEL_ATOMISP)     += atomisp/

Add in Kconfig
source "drivers/staging/media/atomisp/Kconfig"

and after should be visible make menuconfig.

Tuxman2, 2017-03-19:
Do you think it could be a good idea to post a bug report on the kernel bugzilla website for the ov2680, hm2056 and ov5648 support ?

sfumato77, 2017-03-19:
TODO  FILE
gc2235/gc0310/0v2680/ov2722/ov5693/mt9m114...

ov2680 says it's supported. before making a signaling perhaps those who see one of these sensors suportati starch could try ...

Haxk20, 2017-03-19:
Will test it on 4.11 and will see what can I do

Tuxman2, 2017-03-20:
@sfumato77: In my case, I have the Hm2056 and the Ov5648 sensors (front and rear cameras). :-(
I found the driver source code for those 2 sensors (see above) but I don't know if it works. :-(

RussianNeuroMancer, 2017-03-20:
Some changes in UEFI seems like required to get above driver working: https://4pda.ru/forum/index.php?showtopic=650808&view=findpost&p=59557868 

As you can see in long drive can't find right firmware. Where to find it?

sfumato77, 2017-03-20:
in file sh_css_firmware.c

There is talk of firmware
"Irci_stable_candrpv_0415_20150521_0458"

read first byte in shisp_2400b0_v21.bin
http://vavar60.online.fr/share/tablet/chuwi_hi10/Linux_3.14.37/firmware/

which kernel are you using?

maybe it's the driver CLK for baytrail  clk-byt-plt.c but I'm not sure

Tuxman2, 2017-05-26:
Hi,

I made a "bug report" for the support of OV5648 and HM2056 cameras on the kernel bugzilla website. I hope that someone will have a look on it. See it: https://bugzilla.kernel.org/show_bug.cgi?id=195877

dfiloni, 2017-06-11:
ov2680 (4.12rc4) doesn't work on my Lenovo Miix 310 due to atomisp_gmin_platform (same issue posted by @RussianNeuroMancer ).
I ported the ov5680 driver from 01org (uploaded in the bug report) however I cannot test it so I don't think I can ask for its inclusion.

"Failed to find gmin variable" errors can be avoided using a workaround, you can extract vars from DSDT and add them in the code, there is already a struct called hard_vars for this purpose. I think the major issue, at least for me, is the lack of axp regulator support in atomisp_gmin_platform.

Tuxman2, 2017-06-12:
@dfiloni: I talked to Hans de Goede of the portage of OV5648 and HM2056 drivers. He told me that he will looking for when he will have hardware with these cameras. He also told me that atomisp driver don't work at all. I think we have to wait a while before the atomisp driver working.

andy-shev, 2017-10-19:
Guys, I'm pretty sure Sakari, who is de facto maintaining AtomISP driver in upstream, will not take any new code to it without independent Tested-by tag, which effectivelly means we need to get AtomISP working first on the hardware with supported sensors, then convert AtomISP to use V4L2 Sensor APIs (this what Sakari put in update of TODO file recently).

tidus747, 2018-02-08:
@dfiloni I am trying to find a driver for v4l2 for the OV5680 sensor. As you say you have developed a first version that you have not been able to prove. I'm making a platform based on that sensor. If you were so kind to send me the driver I could test it and help you debug it. Thanks in advance !!

Tuxman2, 2018-02-13:
No news about the OV5648 and HM2056 support ?

RussianNeuroMancer, 2018-02-13:
@Tuxman2 
Read whole threadL https://www.mail-archive.com/linux-media@vger.kernel.org/msg123382.html
https://patchwork.kernel.org/project/linux-media/list/?q=atomisp

dfiloni, 2018-02-14:
@tidus747: sorry for the delay, you can find it at https://patchwork.kernel.org/patch/9983661/ . You can also find a patch for HM2056 at https://bugzilla.kernel.org/show_bug.cgi?id=195877 .

tidus747, 2018-02-14:
@dfiloni Thank you very much. As I said, the sensor I'm using is the OV5680. Would I use the OV5648 driver for this sensor? I thought I had read that you had developed the driver for the OV5680.

Tuxman2, 2018-02-15:
@RussianNeuroMancer: Ok. Thanks.
@dfiloni: I can try your new Ov5648 patch but for the HM2056 patch it will not work (see my report in your HM2056 link). In the config file from Hans, the atomisp was disabled. So what options must I enable in the kernel to have the atomisp support (atomisp and mipi camera stuff ?) ?
I found something interesting here: https://github.com/jfwells/linux-asus-t100ta/tree/master/webcam

dfiloni, 2018-02-15:
@tidus747 : I don't think it will work, and 5680 was a typo, I only ported ov5648 and hm2056, sorry for the misunderstanding
@Tuxman2 : I don't know what kernel are you talking about, my patch was for mainline, also probably Hans disabled it because is not supported by your hardware, for example I cannot use ov2680 and hm2056 due to a missing AXP PMIC driver for my laptop (I think). Anyway I don't remember so I cannot help you, sorry

Tuxman2, 2018-02-16:
@dfiloni: In the config file given by Hans and RussianneuroMancer for 4.15.x kernel, the atomisp is disabled by default. So we have to set manually the option to try to make atomisp works. As I seen in the links given by RussianNeuroMancer, the kernel community try to find a solution to make the cameras work. But it seems to not be easy.

andy-shev, 2018-02-16:
Guys, I don't know which language to use to tell you that the driver is (for now) useless. There is no one in the community who has made it working.

Apparently this is a very high demand on testers, but for now even single configuration doesn't work with the upstreamed code.

I recommend to switch to Windows to make camera works. It's pity, but this is unpalatable truth.

Tuxman2, 2018-02-18:
@andy-shev: 

> I recommend to switch to Windows or Android to make camera works. It's pity, but this is unpalatable truth.

Or another solution if we want to use a camera under Gnu/Linux is to buy a small portable and compatible Webcam that can be fixed on the tablet. I don't see others solutions until the cameras in the tablets will be supported.

sonvirgo, 2018-03-22:
I use DroidCam with my NokiaX2sd feeds to Ubuntu Zesty

Tuxman2, 2018-06-14:
No news about the cameras support ?

andy-shev, 2018-06-19:
@Tuxman2
AtomISP v2 driver is gone from upstream. Sorry guys, this must be closed as invalid.
Or if any one can get a **hard** (I mean it!) way by actually extracting all necessary bits from Android...

P.S. Just in case, I'm not a repository owner here, so, whoever they are, its their decision how to proceed now.

Tuxman2, 2018-06-21:
@andy-shev: So the only solution is to buy an external webcam. :(

mirh, 2018-08-16:
OV2680 is now in mainline
https://lwn.net/Articles/747810/

sonvirgo, 2018-08-23:
I adjust matching table
And now the probe is called with error
```
 ~$ dmesg | grep -A3 -B3  OVTI2680
[    6.865235] intel-lpss 0000:00:15.2: enabling device (0000 -> 0002)
[    6.865602] idma64 idma64.2: Found Intel integrated DMA 64-bit
[    6.873799] Enter ov2680_probe()
[    6.873804] ov2680 i2c-OVTI2680:00: xvclk clock missing or invalid
[    6.873868] ov2680: probe of i2c-OVTI2680:00 failed with error -22
[    6.874946] intel-lpss 0000:00:1e.0: enabling device (0000 -> 0002)
[    6.875231] idma64 idma64.3: Found Intel integrated DMA 64-bit
[    6.886305] Intel(R) Wireless WiFi driver for Linux```

andy-shev, 2018-08-23:
@sonvirgo 
Looking at your dmesg output I can tell you that the SoC there provides IPU3. Fortunately there is work in progress to bring it properly to upstream (you may found parts of the driver in vanilla kernel _drivers/media/pci/intel/ipu3/_).  Better to ask about it in linux-media mailing list.

andy-shev, 2018-08-23:
For all who is trying to bring up AtomISP v2, the driver **had been removed** from upstream due to lack of developers who has it working. Just install Windows or supported OS (Android?) to the devices on which you would like to have camera enabled.

Yes, it's pity and especially to me who spend a lot of ours on trying to get something out of it (including a lot of clean up patches pushed in upstream at some point). I share in your pain.

mirh, 2018-08-23:
https://lwn.net/Articles/750572/
@yongzhi1 are you on holiday?

sonvirgo, 2018-08-24:
@andy-shev @mirh Ah now it is clearer, complete suit must cover everything like in Windows. The camera driver alone is not enough but additional driver needed are 
```Intel Control Logic
Intel CSI2 Host Controller
Intel Imaging Signal Processor 2500```
I'll try to test latest vanilla 4.19
Do you think it work now?

andy-shev, 2018-08-24:
@sonvirgo 
I think you need to ask in linux-media mailing list, not here.

Tuxman2, 2018-10-28:
Hi,

@andy-shev: Do you think we can use Anbox software to make the cameras work ?

Thanks.

Tuxman2, 2019-01-22:
Hi, 

any chances to see the cameras work one day or it's definitely dead ?

mirh, 2019-01-22:
IPU3 has been merged into 5.0, for the records.

murlig, 2020-09-12:
I have a baytrail nextbook where i have linux mint 19.3 working fine except the camera. I upgraded the kernel to 5.8.x today. I see the following drivers loaded.
$ lsmod | grep 2680
atomisp_ov2680         32768  0
atomisp_gmin_platform    36864  1 atomisp_ov2680
videodev              245760  1 atomisp_ov2680
mc                     57344  2 videodev,atomisp_ov2680

under /sys
ls /sys/bus/i2c/drivers/ov2680/i2c-OVTI2680\:00
driver  firmware_node  modalias  name  power  subsystem  uevent

dmesg has some possibly worrisome messages?
[   14.052908] atomisp_ov2680: module is from the staging directory, the quality is unknown, you have been warned.
[   14.055714] atomisp_ov2680: module is from the staging directory, the quality is unknown, you have been warned.
[   14.214064] ov2680 i2c-OVTI2680:00: gmin: power management provided via XPower AXP288 PMIC (i2c addr 0x34)
[   14.214089] ov2680 i2c-OVTI2680:00: gmin_subdev_add: ACPI path is \_SB.I2C2.CAM4
[   14.214621] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_ClkSrc
[   14.214628] ov2680 i2c-OVTI2680:00: ClkSrc: using default (1)
[   14.215152] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_CsiPort
[   14.215159] ov2680 i2c-OVTI2680:00: CsiPort: using default (0)
[   14.215680] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_CsiLanes
[   14.215687] ov2680 i2c-OVTI2680:00: CsiLanes: using default (1)
[   14.215867] ov2680 i2c-OVTI2680:00: will handle gpio0 via ACPI
[   14.215974] ov2680 i2c-OVTI2680:00: will handle gpio1 via ACPI
[   14.216069] ov2680 i2c-OVTI2680:00: Failed to find EFI gmin variable gmin_V1P8GPIO
[   14.216077] ov2680 i2c-OVTI2680:00: V1P8GPIO: using default (-1)
[   14.216162] ov2680 i2c-OVTI2680:00: Failed to find EFI gmin variable gmin_V2P8GPIO
[   14.216169] ov2680 i2c-OVTI2680:00: V2P8GPIO: using default (-1)
[   14.216228] ov2680 i2c-OVTI2680:00: Will use CLK1 (pmc_plt_clk_1)
[   14.216755] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_eldo1_1p8v
[   14.216763] ov2680 i2c-OVTI2680:00: eldo1_1p8v: using default (22)
[   14.217588] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_eldo1_sel_reg
[   14.217597] ov2680 i2c-OVTI2680:00: eldo1_sel_reg: using default (25)
[   14.218145] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_eldo1_ctrl_shift
[   14.218152] ov2680 i2c-OVTI2680:00: eldo1_ctrl_shift: using default (0)
[   14.218668] ov2680 i2c-OVTI2680:00: Failed to find EFI variable OVTI2680:00_eldo2_1p8v

ultimately no video when i open cheese.

let me know what to try.

thanks

RussianNeuroMancer, 2020-09-12:
> dmesg has some possibly worrisome messages?

Driver trying to load, fail to do so, but it's not expected to work just yet anyway. So nothing to worry about.

> let me know what to try.

Just wait. Maybe one day it will work on newer kernel (or maybe not)

mirh, 2020-09-12:
Yes, there's still lot of work being in queue
https://git.linuxtv.org/sailus/media_tree.git/log/drivers/staging/media/atomisp

Did anybody check [this](https://github.com/croutor/atomisp2401) btw?

[3 more replies]