burzumishi/linux-baytrail-flexx10 #20: Ubuntu 17.04 and Lenovo miix 3-830 rotation

Haxk20, 2016-11-16:
Today i just installed ubuntu 17.04 alpha on my Lenovo miix 3-830 and rotation of screen work out of the box i think flexx use the same rotation sensor so i think if anybody tries to install it on flexx it should work with ubuntu 17.04 GNOME

nicman23, 2016-11-16:
thanks for reporting it. none had bothered to check it due to other issues. does your talbet have the same hardware?

Haxk20, 2016-11-16:
I think yeah wifi same sound too i get everything from here and working so i think yeah i was like wooow (im using gnome and i see normally in power menu only 3 buttons) but now there were 4 buttons and one of them for rotation lock i already get it to work on 16.10 but it was random then i deleted that install because i messed up my grub but now its working again only its wrong rotated -45 degrees

Haxk20, 2016-11-16:
Only my tablet is mono sound

Haxk20, 2016-11-16:
Still not working sound but its rt5640cr

Haxk20, 2016-11-16:
Even i fixed c-states issue i think by installing linux on 5gb partition its like usb on the motherboard not like partition on the hard drive using it for 2 hours no freeze at all on hard drive 32 gb freeze in 5 minutes or so

Haxk20, 2016-12-02:
Nope i dont fix it what fix Cstates was 4.8 kernel and now on 4.9 i run script which i post issue too

Haxk20, 2016-12-25:
Cstate issues have been fixed by patch or by the script closing this issue

closed 2016-12-25 by Haxk20