burzumishi/linux-baytrail-flexx10 #21: Warning for grub users and UEFI

jharrison022, 2016-11-18:
I'll leave out the gory details about my recent education in UEFI and skip to the point.

(This is relevent to our tablets, as _our_ UEFI isn't completely compatible with efibootmgr.  See below for a tool that is.)

The Debian (at least) grub packages invoke efibootmgr to install an entry for grub into the UEFI NVRAM.  When this occurs, efibootmgr writes a log entry into a mailbox in the NVRAM.  This mailbox will eventually fill up after grub has been updated several times (as is the case with a testing disto.).  efibootmgr isn't properly checking to see if the grub entry exists (on _our_ hardware - its an EFI implementation bug), so it deletes it, writes to the mailbox, then finally writes the grub entry into NVRAM.  You can see where the problem is - the log entry is larger than the boot entry, so grub disappears from the boot menu with no chance of efibootmgr restoring it.

To prevent (or fix this), you should go into /sys/fs/pstore and delete everything there (it is safe to do).

If you find yourself in this state, read this entire page to find out how to get out of it:  https://wiki.archlinux.org/index.php/Unified_Extensible_Firmware_Interface

nicman23, 2016-11-19:
this is a debian bug. We cannot do anything to change that behaviour. Please file a bug on debian tracker.

jharrison022, 2016-11-20:
Just consider it a friendly FYI --

Any distro that uses efibootmgr (that _all_, AFAIK), with either NextBook will get hit with this eventually if you don't monitor the pstore directory.  The NextBook fw doesn't respond correctly to EFI commands which deal with file management of the mailboxes.

Tuxman2, 2016-11-20:
Have you got encounter this message ? 

blockdev: ioctl error on BLKROSET: Operation not permitted

I've got this message each time I make a sudo update-grub.

closed 2016-12-24 by nicman23