burzumishi/linux-baytrail-flexx10 #24: 0001-drivers-staging-rtl8723bs.patch and kernel-4.9-rc5

Tuxman2, 2016-11-21:
Hello,

I tried to applied the 0001-drivers-staging-rtl8723bs.patch to kernel-4.9-rc5 source with the patch -p1 command.

Here is the result  : 

`patching file drivers/staging/Kconfig
Hunk #1 succeeded at 40 with fuzz 2 (offset -2 lines).
patching file drivers/staging/Makefile
Hunk #1 FAILED at 10.
1 out of 1 hunk FAILED -- saving rejects to file drivers/staging/Makefile.rej
patching file drivers/staging/rtl8723bs/Kconfig
patching file drivers/staging/rtl8723bs/Makefile
patching file drivers/staging/rtl8723bs/convert_firmware.c
patching file drivers/staging/rtl8723bs/convert_firmware.h
patching file drivers/staging/rtl8723bs/core/rtw_ap.c
patching file drivers/staging/rtl8723bs/core/rtw_btcoex.c
patching file drivers/staging/rtl8723bs/core/rtw_cmd.c
patching file drivers/staging/rtl8723bs/core/rtw_debug.c
patching file drivers/staging/rtl8723bs/core/rtw_eeprom.c
patching file drivers/staging/rtl8723bs/core/rtw_efuse.c
patching file drivers/staging/rtl8723bs/core/rtw_ieee80211.c
patching file drivers/staging/rtl8723bs/core/rtw_io.c
patching file drivers/staging/rtl8723bs/core/rtw_ioctl_set.c
patching file drivers/staging/rtl8723bs/core/rtw_mlme.c
patching file drivers/staging/rtl8723bs/core/rtw_mlme_ext.c
patching file drivers/staging/rtl8723bs/core/rtw_odm.c
patching file drivers/staging/rtl8723bs/core/rtw_pwrctrl.c
patching file drivers/staging/rtl8723bs/core/rtw_recv.c
patching file drivers/staging/rtl8723bs/core/rtw_rf.c
patching file drivers/staging/rtl8723bs/core/rtw_security.c
patching file drivers/staging/rtl8723bs/core/rtw_sta_mgt.c
patching file drivers/staging/rtl8723bs/core/rtw_wlan_util.c
patching file drivers/staging/rtl8723bs/core/rtw_xmit.c
patching file drivers/staging/rtl8723bs/hal/Hal8723BPwrSeq.c
patching file drivers/staging/rtl8723bs/hal/Hal8723BReg.h
patching file drivers/staging/rtl8723bs/hal/HalBtc8723b1Ant.c
patching file drivers/staging/rtl8723bs/hal/HalBtc8723b1Ant.h
patching file drivers/staging/rtl8723bs/hal/HalBtc8723b2Ant.c
patching file drivers/staging/rtl8723bs/hal/HalBtc8723b2Ant.h
patching file drivers/staging/rtl8723bs/hal/HalBtcOutSrc.h
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_BB.c
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_BB.h
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_MAC.c
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_MAC.h
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_RF.c
patching file drivers/staging/rtl8723bs/hal/HalHWImg8723B_RF.h
patching file drivers/staging/rtl8723bs/hal/HalPhyRf.c
patching file drivers/staging/rtl8723bs/hal/HalPhyRf.h
patching file drivers/staging/rtl8723bs/hal/HalPhyRf_8723B.c
patching file drivers/staging/rtl8723bs/hal/HalPhyRf_8723B.h
patching file drivers/staging/rtl8723bs/hal/HalPwrSeqCmd.c
patching file drivers/staging/rtl8723bs/hal/Mp_Precomp.h
patching file drivers/staging/rtl8723bs/hal/hal_btcoex.c
patching file drivers/staging/rtl8723bs/hal/hal_com.c
patching file drivers/staging/rtl8723bs/hal/hal_com_phycfg.c
patching file drivers/staging/rtl8723bs/hal/hal_intf.c
patching file drivers/staging/rtl8723bs/hal/hal_phy.c
patching file drivers/staging/rtl8723bs/hal/hal_sdio.c
patching file drivers/staging/rtl8723bs/hal/odm.c
patching file drivers/staging/rtl8723bs/hal/odm.h
patching file drivers/staging/rtl8723bs/hal/odm_AntDiv.c
patching file drivers/staging/rtl8723bs/hal/odm_AntDiv.h
patching file drivers/staging/rtl8723bs/hal/odm_CfoTracking.c
patching file drivers/staging/rtl8723bs/hal/odm_CfoTracking.h
patching file drivers/staging/rtl8723bs/hal/odm_DIG.c
patching file drivers/staging/rtl8723bs/hal/odm_DIG.h
patching file drivers/staging/rtl8723bs/hal/odm_DynamicBBPowerSaving.c
patching file drivers/staging/rtl8723bs/hal/odm_DynamicBBPowerSaving.h
patching file drivers/staging/rtl8723bs/hal/odm_DynamicTxPower.c
[... 203 lines cut ...]
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
4 out of 4 hunks ignored -- saving rejects to file sound/soc/intel/skylake/skl-topology.c.rej
patching file sound/soc/intel/skylake/skl-topology.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file sound/soc/intel/skylake/skl-topology.h.rej
patching file sound/soc/intel/skylake/skl-tplg-interface.h
Hunk #1 FAILED at 210.
Hunk #2 succeeded at 179 with fuzz 2 (offset -49 lines).
1 out of 2 hunks FAILED -- saving rejects to file sound/soc/intel/skylake/skl-tplg-interface.h.rej
patching file sound/soc/intel/skylake/skl.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file sound/soc/intel/skylake/skl.c.rej
patching file sound/soc/intel/skylake/skl.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file sound/soc/intel/skylake/skl.h.rej
`
I think if I compile this kernel with this patch I will get an error during compilation.

jharrison022, 2016-12-26:
Eh.  Don't bother with kernel patches.  Just use the driver from here:  https://github.com/hadess/rtl8723bs

nicman23, 2016-12-28:
also the wifi works only with g nets irrc (or it just doesn't work with 300n which is what i have)

closed 2016-12-28 by nicman23