burzumishi/linux-baytrail-flexx10 #25: C-state fix

Haxk20, 2016-12-02:
Hi i was having the C-state issue all the time even the fix on this repository dont help but until i find this page: https://forum.manjaro.org/t/intel-bay-trail-freezes-the-linux-kernel/1931/3 and you need the script: Disable all C6 states enable all C7 core states for Baytrail CPUs from this site:     https://bugzilla.kernel.org/show_bug.cgi?id=109051 and after making the script service its working without making the tablet warm like if it was in oven and battery life is even better.
IMPORTANT: you need to remove the "intel_idle.max_cstate=1" from kernel parameter and even if you have anything else in kernel parameter what is connected with C-states and modifing them in some way you need to remove this too otherwise it will not work.
VERY BIG THANKS TO: Wolfgang M. Reimer for creating this script.

Tuxman2, 2016-12-18:
Hi,

@Haxk20: Do you think we can use your tip for Atom Z3735F processor, or is there a risk  ?

Thanks.

Haxk20, 2016-12-22:
Its not in the list but its working totally awesome i have the Z3735F too and no freeze for 2 months

Tuxman2, 2016-12-22:
4.9.x kernel seems to correct the c-state problem but 4.9.x have some regressions. Do we still need the intel_pstates=disabled options ?

Haxk20, 2016-12-22:
4.9 didnt fix it for me at all it make it to stay alive longer time but froze too and dont know about the pstates in 4.9 but with this script you dont need it with this script you dont need any modification to cstates or pstates its fix them but its need to the script needs to run to work so make it to start at boot and no kernel parameter for cstates and pstates and you will not have any freeze

Tuxman2, 2016-12-23:
It seems that someone create a patch to avoid using the script that you tell. 

See it: https://bugzilla.kernel.org/show_bug.cgi?id=109051#c609 and it:  https://bugzilla.kernel.org/show_bug.cgi?id=109051#c610

Haxk20, 2016-12-23:
Yes but I prefer script because i dont compile kernel too often because arch is rolling distro so update will update kernel too so then i need to compile it again for the new kernel so i prefer script

nicman23, 2016-12-23:
@Haxk20 do you know the magic that is ccache?

Haxk20, 2016-12-23:
Wow didnt know about ccache its cool how much time it can save ?

nicman23, 2016-12-23:
depends on the hdd / sdd . mine is about a third(?) but i have not 
really timed it.

Haxk20, 2016-12-24:
Wow thats awesome can i compile kernel from Rpi with arch linux for my tablet running arch too ?

Haxk20, 2016-12-24:
Rpi is ARM and tablet is x86_64 (like normal 64 bit desktop PC) is possible to compile from arm to x86_64 ?

nicman23, 2016-12-24:
You first ask yourself if you want 2 days to wait around for the compiler to finish

Haxk20, 2016-12-24:
Yeah well thats right rpi is slow well i think i will make some free space in my arch desktop how much free space do i need ?

nicman23, 2016-12-25:
About 8g

Haxk20, 2016-12-25:
But i mean even with source ? Or u already count source files to it ?

Haxk20, 2016-12-25:
And even 1 little question i use wifi usb adapter because on ubuntu rtl8723bs make the tablet to freeze (it was not cstate issue) is rtl8723bs freezing on Arch too ?

nicman23, 2016-12-25:
Yes, the driver is not stable -for me anyways

Haxk20, 2016-12-25:
Is there some patch for the wifi or i need to use usb adapter

nicman23, 2016-12-25:
not yet

beevabeeva, 2016-12-25:
Have you tried Linuxium's images/patches? (They're all Deb based so not the way to go if you're dead set on Arch but maybe a good place to start?)

Check http://www.linuxium.com.au/how-tos to see what Ian Morrison did and/or get the ready-to-boot images.

With his Lubuntu image, wifi works OTB (Nextbook 10.1 - *not*  flexx- Z3735G processor with RTL8723BS and BTrail chipset...not sure the differences with the flexx ). Other images should work the same.

 Video is buggy OTB (for me). Playing with C-state and swappiness seemed to improve it. Still not as good as stock on Windows (just to give a benchmark...).

Reboot and shutdown hangs, have to hard shutdown with power button for now.

Sound does not work. Tried Ian's solution as he described - to no avail. Have not had time to look into it further.

Battery indicator shows full all the time.

Screen brightness not working(have not tried the fixes discussed here).

Keyboard works.

Bluetooth - can't detect anything or be detected so it's not working (for me anyway). Have not looked into it yet.

Rotation and multitouch don't work but there is another image:
http://xjesus.net/xjubuntu/xjubuntab-tablet-customizedcustomizable-distro-based-on-xubuntu-14-04-64-bits/
Although on this one, the wifi hasn't been patched. Don't think anything else works.

Sorry in advance, this is probably not the right place to put this: Thanks to burzumishi and everyone here contributing to this project. It's really cool to watch open source development in motion, with such talent.
Still doing CS/math degree and learning kernel but hopefully will be able to start contributing here soon.

PS - not asking for help in this message just sharing what I've done in case it is of any benefit to anyone.

youling257, 2016-12-25:
intel_idle.max_cstate=1 ，c6off-c7on.sh no use .

x86:/ $ su
x86:/ # sush /storage/A4AE40E7AE40B398/ADM/cstateInfo.sh
cpu0 State  Name    Disabled  Latency  Residency      Time   Usage
         0  POLL           0        0          0    165188     541
         1  C1-BYT         0        1          1  66937082  167920
cpu1 State  Name    Disabled  Latency  Residency      Time   Usage
         0  POLL           0        0          0    132716     263
         1  C1-BYT         0        1          1  72517489   80255
cpu2 State  Name    Disabled  Latency  Residency      Time   Usage
         0  POLL           0        0          0    127568     172
         1  C1-BYT         0        1          1  72318482   68787
cpu3 State  Name    Disabled  Latency  Residency      Time   Usage
         0  POLL           0        0          0    112984     166
         1  C1-BYT         0        1          1  74612115   77764
x86:/ #

intel_idle.max_cstate=2，c6off-c7on.sh just for C6N-BYT

x86:/ $ su
x86:/ # sush /storage/A4AE40E7AE40B398/ADM/cstateInfo.sh
cpu0 State  Name     Disabled  Latency  Residency       Time   Usage
         0  POLL            0        0          0     209967     372
         1  C1-BYT          0        1          1  119676918  244533
         2  C6N-BYT         1      300        275    9892356    8022
cpu1 State  Name     Disabled  Latency  Residency       Time   Usage
         0  POLL            0        0          0     172986     345
         1  C1-BYT          0        1          1  126686914  114641
         2  C6N-BYT         1      300        275   10381405    5841
cpu2 State  Name     Disabled  Latency  Residency       Time   Usage
         0  POLL            0        0          0     211053     317
         1  C1-BYT          0        1          1  124165824  103528
         2  C6N-BYT         1      300        275    9802324    2413
cpu3 State  Name     Disabled  Latency  Residency       Time   Usage
         0  POLL            0        0          0     229982     277
         1  C1-BYT          0        1          1  126706192  117251
         2  C6N-BYT         1      300        275   11696947    3066
x86:/ #

Haxk20, 2016-12-26:
well i compiled the kernel today BT is working with the rtl8723bs_bt driver volume buttons are working with patches still no battery indicator but i dont applied the patches for it and why u use those ugly kernel parameters when u have this script and even guy created a patch in the forum above in my post sound is still not working but i will apply the new firmware UCM files and asound then i hope it will work clocksource jiffies is happy now with that configuration and its running ok will test tsc too will post here what it does

Haxk20, 2017-01-03:
Tsc still have problems but i read on forum that it cannot find the right frequency for the processor then it will start to do crazy things

Haxk20, 2017-01-06:
Can somebody add this fix to README on this repo this can help a lot of people.

closed 2017-01-06 by Haxk20