burzumishi/linux-baytrail-flexx10 #26: Arch on this device

Haxk20, 2016-12-02:
Hi anyone mabaged to go for Arch linux on this device i installed it but after that i installed ubuntu 17.04 back because of that: touchscreen act very weird when i move my finger on desktop it give the selection like thing you know and sometimes its goes off which is good and sometimes its stays on the screen on ubuntu its working awesome even multitouch support without touchegg is there some way to take the ubuntu drivers and configs and other thing and load it into the arch or i should stay at ubuntu ?

nicman23, 2016-12-18:
Annoy me again in 2 weeks , I ll update the pkgbuilds. They work btw if you don't care about an old kernel. https://github.com/nicman23/misc_aur

Haxk20, 2016-12-23:
Ok i got Arch working litlle bit better by using jiffies clocksource but it slows down or speed up after sometime so unusable after time and need restart trying to find perfect clocksource but cannot find it

nicman23, 2016-12-24:
move to #28 and for the pkgbuilds if i ever find the time ~~or the charger...~~ to my repo

closed 2016-12-24 by nicman23