burzumishi/linux-baytrail-flexx10 #28: What clocksource do you use on this device ?

Haxk20, 2016-12-22:
Title says it what clocksource do you use or is there some patch which will fix the problems with the clocksource ?

Tuxman2, 2016-12-22:
Hi,

clocksource for baytrail audio ?

Haxk20, 2016-12-22:
No for system

Haxk20, 2016-12-22:
But you can even say the baytrail clocksource. But if you dont know how to check what CS you use for system do cat /sys/devices/system/clocksource/clocksource0/current_clocksource and post resoult here please and then type cat /sys/devices/system/clocksource/clocksource0/available_clocksource and post resoult here too please

jharrison022, 2016-12-23:
tsc.  What issues are you seeing?

Haxk20, 2016-12-23:
With tsc the system is so fast and unexpected tx usb frame happens lot of time so when i type anything it type it 100 times and system clock is like 5 times faster sometime and at boot black screen when i move the cursor after 2 minutes its work ok but still when i type anything i type it 10 times and system is so fast sometime can you help me with this ?

nicman23, 2016-12-24:
are you using any clock patches? i had to for audio..

Haxk20, 2016-12-24:
Yeah using jiffies for now and no i use kernel on arch with no patches at all what do you recommend ? Will the baytrail clocksource patch fix my problem ?

Haxk20, 2016-12-24:
Because i want to make the tablet usable but the clocksource is not working perfect everybody is saying that on baytrail tsc works  good but for me R-jiffies is ok but slows down, tsc boots ok login manager is ok but after login black screen when i move mouse after 2 minutes its started to work ok then after 3 minutes unexpected tx usb frame happens and its unusable again because its so fast, jiffies is for now best but still slows down but not that fast as R-jiffies do. So do you have any idea that can help me ? Please i dont know what to do and even i dont know if its on 4.9 kernel or where it was but i saw acpi_pm clocksource but dont see it on arch now acpi_pm was kinda ok but speeds up 3 times for about 4 seconds but thats normal on those clocksources but is there some patch to make some of those clocksources work normally as they should ?

Haxk20, 2017-01-01:
hello do you use any patch to make tsc to work because when i use jiffies wifi rtl8723bs is not working because its too slow and cannot find any wifi access point but when i use tsc it finds it and working but TSC is just buggy for me any patch to make some clocksource to work normally ?

Haxk20, 2017-01-06:
Well closing still dont fixed the problem but whatever i will try what i can.

closed 2017-01-06 by Haxk20