burzumishi/linux-baytrail-flexx10 #29: Ext4 ot Btrfs file system for Bay Trail tablet ?

Tuxman2, 2016-12-22:
Hello everybody,

I only want to know what is the best file system for Bay Trail tablet ? Ext4 or Btrfs ?
Btrfs have snapshots, ext4 not (ext4 is journalised).

If the Baytrail tablet have Windows OS, is it a good idea to create only a single root partition. Or, must we also create a /home and swap partition on the Emmc disk ?

Thank you for your advice.

Haxk20, 2016-12-22:
I use ext4 all time and its good

Tuxman2, 2016-12-23:
Ok, and what do you think about a single root partition ? Separate /home and swap are necessary on EMMC disk ?

Haxk20, 2016-12-23:
No i have everything in one partition and 2 gb Swap and its ok but you can create separate home partition if you want

Tuxman2, 2016-12-24:
I think swap partition is not a good thing for SSD disk or EMMC disk. SSD and EMMC are flash memory. In my case, I don't create a swap partition.

nicman23, 2016-12-24:
btrfs is not there yet and there is no advantage in such small drive capacity.

also there is no need for swap in a 2gb tablet.

SSD and other flash media are not fragile enough to care about swap.

The real question is if we should be running f2fs. From my test the answer is yes. However, there is a "bug" (not really it is upstream behaviour) that every time you boot with a new kernel, the drive will run a full fcsk.

Ask anything else you might want and then please close the issue :)

Haxk20, 2016-12-24:
@nicman23 look at studioteabahlg site the guy on this site have baytrail-M tablet running arch with f2fs i think he say something about it but i dont read it to end because i dont use f2fs

nicman23, 2016-12-24:
I do, it works except for the fsck thing

Haxk20, 2016-12-24:
What is fsck ? If i can ask

nicman23, 2016-12-25:
Filesystem check

Haxk20, 2016-12-25:
Oh ok but ext4 do that too sometimes but most if i turn it off hard

Tuxman2, 2016-12-25:
Ok, I will close this issue.

closed 2016-12-25 by Tuxman2