burzumishi/linux-baytrail-flexx10 #30: problem with Ubuntu 64 bit version and Grub

Tuxman2, 2016-12-30:
Hello,

before, I used a 32 bit linux version (Ubuntu), now I installed a 64 bit linux version of Ubuntu. But I've got a big problem, I can't start Ubuntu 64 bit version through grub menu because I've got a black screen and nothing happens. Grub-efi-ia32 and grub-efi-ia32-bin are installed. I've got grubia32.efi in /EFI/Ubuntu directory. Ubuntu is listed in bios entries and it is available with efibootmgr command. I can boot on my Ubuntu system if I use the Live-DVD and the "c" command of Grub. Then I launch these commands through the grub prompt and it works :

grub > linux (hd1,gpt5)/boot/vmlinuz-4.4.0-xxx-generic root=/dev/mmcblk0p5 ro quiet splash
grub > initrd (hd1,gpt5)/boot/initrd.img-4.4.x-xxx-generic
grub > boot

I tried to defined the fist line (linux (hd1,gpt5)...) in the /etc/default/grub -> GRUB_CMDLINE_LINUX but when I boot I've got an error message. I used sudo update-grub

How to solve my problem ? Have got an idea ?

Note: grub-efi-ia32 was installed during the installation of the distribution, but I reinstalled it after to be sure that grub-efi-ia32 was OK.

Thanks for your help.

nicman23, 2016-12-30:
try move the grub.efi (or whatever it is called) to 
$esp/EFI/boot/bootia32.efi iirc

this would in a windows installation be the default thing that the 
machine would boot into. these cheapo tablets do not follow the uefi 
spec and will not boot to what their efivars point them too.

there is a section in the arch wiki on the uefi article about the same 
issue on x64 uefi and the machine i am typing this has the exact same 
problem (thanks hp)

Haxk20, 2016-12-30:
Mount efi partition to /boot/efi then Do $ sudo grub-install --target=i386-efi --efi-directory=/boot/efi --bootloader-id=Ubuntu --recheck that should install grub as 32bit version and it should start

Haxk20, 2016-12-30:
That needs to be done in the installed os not in live dvd or usb because it will not work the problem with this is that the 64 bit version install 64 bit grub but we need 32 bit grub because we have 32 bit uefi the point is that 32 bit os install 32 bit grub thats why you can easily boot into 32 bit os the solution is mounting EFI partition into /boot/efi and doing that command it will install 32 bit grub and os will hopefully boot

Tuxman2, 2017-01-01:
Thanks to nicman23 and Haxk20 for your replies and Happy New Year.  :)

Do I need to define "linux (hd1,gpt5)/boot/vmlinuz-4.4.0-xxx-generic root=/dev/mmcblk0p5 ro quiet splash" in the /etc/default/grub -> GRUB_CMDLINE_LINUX_DEFAULT line after that ?

nicman23, 2017-01-01:
Please run update-grub2 and post the contents of your EFI partition

Tuxman2, 2017-01-01:
@nicman23: I will try the solution of Haxk20 later because for now I'm not on my hybrid machine. I will post the results later.

Haxk20, 2017-01-01:
yeah you need to run update-grub to make grub.cfg to find all the partitions or if you want to use longer command you can: grub-mkconfig /boot/grub/grub.cfg this one will make sure its in correct path

Tuxman2, 2017-01-03:
Hello,

I tried the solution of Haxk20 but I always got the same problem.

1.  $ sudo grub-install --target=i386-efi --efi-directory=/boot/efi --bootloader-id=Ubuntu --recheck
Result:
plaform for i386-efi is installed and without error.

2. sudo update-grub
Result:
linux image found : /boot/vmlinuz-4.4.0-53-generic
initial memory found: /boot/initrd.img-4.4.0-53-generic
Windows boot manager found on /dev/mmcblk0p1@EFI/Microsoft/boot/bootmgfw.efi
Adding boot menu entry for EFI firwmare configuration
Done.

3. Boot entry in grub: linux /boot/vmlinuz-4.4.0-53-generic root=UUID=3ea2e6a-9742-4771-a769-a84087d9c148  quiet splash intel_idle.max_cstate=0 intel_pstates=disabled

4.  /etc/default/grub:
    GRUB_CMDLINE_LINUX_DEFAULT="quiet splash intel_idle.max_cstate=0 intel_pstates=disabled"
    GRUB_CMDLINE_LINUX=""

5. /etc/fstab (for mmcblk0p5): root=UUID=3ea2e6a-9742-4771-a769-a84087d9c148  /  ext4  errors=remount-ro 0       1

6.  Live-USB and the "c" command of Grub. I launch these commands through the grub prompt.

grub > linux (hd1,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk0p5 intel_idle.max_cstate=0 intel_pstates=disabled
grub > initrd (hd1,gpt5)/boot/initrd.img-4.4.0-53-generic
grub > boot
Result: It works.

7. Launch the first entry through grub menu without live-usb.
Result: Blackscreen and nothing happens.

8. I defined this line in grub entry with the 'e' command: linux (hd1,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk0p5 intel_idle.max_cstate=0 intel_pstates=disabled
Result:
Error: That partition doesn't exist
unaligned point 0x78b1479
Aborted

9. I tried the recovery mode
Result:
Boot stops after loading initial memory image and initial ramdisk messages.

I don't know what can I do. Thanks if you find the solution. ;-)

Haxk20, 2017-01-03:
Its the same problem that i was having in arch linux DONT USE UUID.
The error says it Error: That partition doesn't exist grub cannot find the partition so you are giving grub wrong location of kernel try
1.  linux (hd0,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk0p5 intel_idle.max_cstate=0 intel_pstates=disabled
2.  linux (hd0,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk1p5 intel_idle.max_cstate=0 intel_pstates=disabled
3.  linux (hd1,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk1p5 intel_idle.max_cstate=0 intel_pstates=disabled
problem is that th live usb grub see the hard disk different then installed grub i dont know why but its true so when you copy kernel command and initrd from live usb it will not fit normal grub.
best will be if you boot grub without live-usb and press "c" to enter command line and type "ls" and post the output so i can see how your drive partition list looks like and give you the right commands to enter

nicman23, 2017-01-03:
or you know.... `ls`

Haxk20, 2017-01-03:
@nicman23 ? I got something wrong typed ?

Tuxman2, 2017-01-04:
Thanks.

If I want to use that line permanently,  I guess I must put this line in /etc/default/grub -> GRUB_CMDLINE_LINUX and disable GRUB_CMDLINE_LINUX_DEFAULT ?

Haxk20, 2017-01-04:
I know its strange but boot into uefi boot and select boot option ubuntu without the USB plugged in !! Then type the commands what they work when this will boot then the problem is in your uefi i got this problem was the same as your first error the partition didnt exist and when i give grub the right path only blackscreen then when i almost gave up i turned tablet off booted into uefi and selected ubuntu from the entry list you must have the ubuntu entry in there for sure its in the location when you try to boot usb then there will be Ubuntu and again dont have usb plugged in after you select it everything will look the same but now when you press c for commsnd line type the command which works and you should boot into ubuntu if not then i dont know i dont own your tablet mine is lenovo miix 3-830 with almost the same hardware as flexx but different ssd touchscreen and few things so i cannot know whats wrong but im sure at 80% that botting from uefi the ubuntu then type the commands should work post result of it here ;)

Haxk20, 2017-01-04:
And if i can ask why do you use 4.4 kernel and not 4.9 ? 4.9 have better support for our devices and the problem with blackscreen boot disappear in 4.8 before 4.8 you must boot into uefi and from there boot the ubuntu entry then type commands then it will finally boot but 4.8 fix that for me 100%

Haxk20, 2017-01-04:
Of the time

Tuxman2, 2017-01-04:
@Haxk20: ok, as you said I boot into uefi boot menu and I choose Ubuntu entry. I press 'c' command and I use the right commands. Right? Must I use ?: 
linux (hd1,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk0p5 intel_idle.max_cstate=0 intel_pstates=disabled 
 initrd (hd1,gpt5)/boot/initrd.img-4.4.0-53-generic
or 
linux (hd0,gpt5)/boot/vmlinuz-4.4.0-53-generic root=/dev/mmcblk0p5 intel_idle.max_cstate=0 intel_pstates=disabled
 initrd (hd0,gpt5)/boot/initrd.img-4.4.0-53-generic

The kernel available by default is 4.4.0-53. I will install the 4.8.0-32 kernel after. Why the 4.4 kernel have this problem and not the 4.8 ?

nicman23, 2017-01-04:
Please, stop spamming. The grub command line has a lot of applets that you can use like ls, cd and configload. navigate to the efi partition and use auto-competition with the tab key to help you run the correct kernel and ramdisk (initrd)

Haxk20, 2017-01-05:
@tuxman2 try both and see with one work and can you boot now without live-usb ? Or still cannot boot can you boot to ubuntu when you boot first into uefi and boot ubuntu like that ?

Tuxman2, 2017-01-05:
@nicman23: Sorry for spamming.
@Haxk20: I tried what you said yesterday and it worked. :-D I installed 4.8.0-xx kernel from Ubuntu and it fixed the problem. Thank you very much man. ;-)
I don't know why 4.4.0-xx kernel have this problem and not the 4.8.0-xx kernel.

I close this topic but I have others issues. I will post these one in the other topics.

closed 2017-01-05 by Tuxman2

Haxk20, 2017-01-06:
Kernel 4.8 didnt have this because on 4.8 some guy started to mainline some patches for this and poweroff and reboot and other stuff 4.9 added even more working stuff 4.10 too and 4.11 will probably what i have read add support for battery percentage out of the box ;)