burzumishi/linux-baytrail-flexx10 #31: Errors from dmesg command

Tuxman2, 2017-01-05:
Hello, 
Here is the errors I've got with dmesg command (kernel 4.8.0-xx):

[    0.000000] efi: requested map not found.
[    0.000000] esrt: ESRT header is not in the memory map.

 0.264227] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.264256] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.264280] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9f4bb68bb848), AE_NOT_EXIST (20160422/psparse-542)
[    0.264368] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.264393] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.264415] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9f4bb68bb848), AE_NOT_EXIST (20160422/psparse-542)
[    0.264449] ACPI Error: Method execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9f4bb68bb848), AE_NOT_EXIST (20160422/uteval-103)
[    0.264593] ACPI: Power Resource [P28P] (off)
[    0.264729] ACPI: Power Resource [P18P] (off)
[    0.265215] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.265242] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.265265] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9f4bb68bb410), AE_NOT_EXIST (20160422/psparse-542)
[    0.265347] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.265371] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.265393] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9f4bb68bb410), AE_NOT_EXIST (20160422/psparse-542)
[    0.265426] ACPI Error: Method execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9f4bb68bb410), AE_NOT_EXIST (20160422/uteval-103)
[    0.265840] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.265866] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.265889] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9f4bb68bb4b0), AE_NOT_EXIST (20160422/psparse-542)
[    0.265971] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.265995] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.266017] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9f4bb68bb4b0), AE_NOT_EXIST (20160422/psparse-542)
[    0.266050] ACPI Error: Method execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9f4bb68bb4b0), AE_NOT_EXIST (20160422/uteval-103)
[    0.266541] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.266567] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.266590] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9f4bb68bb9d8), AE_NOT_EXIST (20160422/psparse-542)
[    0.266671] ACPI Error: No handler for Region [GPOP] (ffff9f4bb68b6120) [GeneralPurposeIo] (20160422/evregion-166)
[    0.266696] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.266718] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9f4bb68bb9d8), AE_NOT_EXIST (20160422/psparse-542)
[    0.266751] ACPI Error: Method execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9f4bb68bb9d8), AE_NOT_EXIST (20160422/uteval-103)

8.413768] i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed

9.137847] genirq: Flags mismatch irq 8. 00000083 (volume_down) vs. 00000080 (rtc0)
[    9.140450] gpio-keys gpio-keys.0.auto: Unable to claim irq 8; error -16
[    9.146267] gpio-keys: probe of gpio-keys.0.auto failed with error -16
 
[  165.187468] EXT4-fs (mmcblk1p2): VFS: Can't find ext4 filesystem
[  165.197232] EXT4-fs (mmcblk1p2): VFS: Can't find ext4 filesystem
[  165.205204] EXT4-fs (mmcblk1p2): VFS: Can't find ext4 filesystem
[  165.212373] squashfs: SQUASHFS error: Can't find a SQUASHFS superblock on mmcblk1p2
[  165.220224] FAT-fs (mmcblk1p2): bogus number of reserved sectors
[  165.220243] FAT-fs (mmcblk1p2): Can't find a valid FAT filesystem
[  165.255574] XFS (mmcblk1p2): Invalid superblock magic number
[  165.267612] FAT-fs (mmcblk1p2): bogus number of reserved sectors
[  165.267624] FAT-fs (mmcblk1p2): Can't find a valid FAT filesystem
[  165.283459] VFS: Can't find a Minix filesystem V1 | V2 | V3 on device mmcblk1p2.
[  165.295865] hfsplus: unable to find HFS+ superblock
[  165.301014] qnx4: no qnx4 filesystem (no root dir).
[  165.309428] ufs: You didn't specify the type of your ufs filesystem
               
               mount -t ufs -o ufstype=sun|sunx86|44bsd|ufs2|5xbsd|old|hp|nextstep|nextstep-cd|openstep ...
               
               >>>WARNING<<< Wrong ufstype may corrupt your filesystem, default is ufstype=old
[  165.309951] ufs: ufs_fill_super(): bad magic number
[  165.317386] hfs: can't find a HFS filesystem on dev mmcblk1p2
[  165.504806] EXT4-fs (mmcblk1p3): VFS: Can't find ext4 filesystem
[  165.510640] EXT4-fs (mmcblk1p3): VFS: Can't find ext4 filesystem
[  165.515557] EXT4-fs (mmcblk1p3): VFS: Can't find ext4 filesystem
[  165.520375] squashfs: SQUASHFS error: Can't find a SQUASHFS superblock on mmcblk1p3
[  165.524667] FAT-fs (mmcblk1p3): bogus number of reserved sectors
[  165.524677] FAT-fs (mmcblk1p3): Can't find a valid FAT filesystem
[  165.546419] XFS (mmcblk1p3): Invalid superblock magic number
[  165.557581] FAT-fs (mmcblk1p3): bogus number of reserved sectors
[  165.557592] FAT-fs (mmcblk1p3): Can't find a valid FAT filesystem
[  165.571092] VFS: Can't find a Minix filesystem V1 | V2 | V3 on device mmcblk1p3.
[  165.575244] hfsplus: unable to find HFS+ superblock
[  165.579287] qnx4: no qnx4 filesystem (no root dir).
[  165.583022] ufs: You didn't specify the type of your ufs filesystem
               
               mount -t ufs -o ufstype=sun|sunx86|44bsd|ufs2|5xbsd|old|hp|nextstep|nextstep-cd|openstep ...
               
               >>>WARNING<<< Wrong ufstype may corrupt your filesystem, default is ufstype=old
[  165.583408] ufs: ufs_fill_super(): bad magic number
[  165.587510] hfs: can't find a HFS filesystem on dev mmcblk1p3

Are there any solutions to correct these errors ? For the last problem, do you think it can be the solution, or is it another problem ? :

4.8.0 Workaround (Bug #150881)

This makes sure the root filesystem on a MicroSD card can be found.
	
Device Drivers
⌞ Real Time Clock
⌞ PC-Style 'CMOS'

CONFIG_RTC_DRV_CMOS [=m]

Thanks.

Haxk20, 2017-01-06:
Im happy that i helped you fix the grub problem but to fix as most of the errors here you need to apply patches dont know what patches but if you can apply patches for acpi this will fix some of acpi errors.

And if i can ask what are those Real Time Clock and the PC-Style 'CMOS' what they fix ?

Tuxman2, 2017-01-06:
@Haxk20: thanks for your help about my grub problem. ;-)
You talked about patches for acpi, is it the patches available in the kernel directory from burzumishi github ? I also have errors for the Baytrail sound card (I posted a message in the sound audio problem topic).
For Real Time Clock and the PC-Style 'CMOS' problem, here is what I found o nthe net:
There is a small regression in the current Kernel release that can cause the MicroSD slot to not function and there by not allowing the root file system to be mounted. The "fix" for this is trivial and is already in place in the config file. The workaround is to set the RTC PC CMOS clock as a module.

Thanks.

Tuxman2, 2017-01-08:
Hi,

Is someone here has tested/compiled the custom kernel sources from Linuxium for Ubuntu and derivates and perhaps the arch distribution (see Linuxium blog and github here: https://github.com/linuxium?tab=repositories) ? Do you think that custom kernel sources from linuxium needs more patches (axp288 for battery, sound, backlight, i915 patches from here, ...) ?

beevabeeva, 2017-01-08:
Yes I've tested it and posted details in an earlier email.

Tuxman2, 2017-01-09:
@beevabeeva: OK. Are they safe (I mean by that if it can break the hardware of the computer) ?
These kernels solve some problems (sound, c-state, wifi, bluetooth). Mainly with the kernel 4.9.0-11.12 offered. I guess these kernels don't include the patches for axp288 problem, and the patches for i915 available here from burzumishi. Thanks if you can post details on it.
Do you try to compile the custom kernel sources from Linuxium ?

Tuxman2, 2017-01-12:
@beevabeeva: No news about your tests on Linuxium custom kernels ?

Someone else has tested these custom kernels ?

Thanks.

beevabeeva, 2017-01-13:
Sorry have exams till 19th then will tell all

Tuxman2, 2017-01-15:
@beevabeeva: Ok. It seems that we must be carefull with the custom kernels from Linuxium and the Linuxium isos. They include experimental drivers and microprogram who can break the hardware. Then I don't know if it is a good idea to use them. Perhaps it is more safe to follow the informations from here by applying the patches recommended in the different messages to make custom kernels.

closed 2017-02-20 by Tuxman2