burzumishi/linux-baytrail-flexx10 #33: Weird problem with tsc

Haxk20, 2017-01-15:
I finally somehow fixed few problems with tsc clocksource still not perfect but better then it was but now i have this problem tsc runs great in gdm or in CLI but not in GUI its laggy and not great at all i cant even watch yt videos on it because its laggy i saw that laggy problem when cpu frequency drop then it lag and when it goes up its ok again but its laggy so much. I thought that you guys should have some idea how to fix this maybe changing something in i915 on in cpu profile or anything else even when you know how to fix it and there is patch post it here i can build kernel and test Thanks.

Tuxman2, 2017-01-15:
Hi Haxk20,

I would like to know what is tsc and tsc clocksource. Some people put tsc in grub command line. Is there a relationship with c-state, or can be with wifi?

Thanks.

Haxk20, 2017-01-16:
Tsc is clocksource how cpu ticks and what exactly they put in grub command line ?

Haxk20, 2017-01-16:
I put tsc=reliable and its again terrible with clocksource=tsc its kinda ok almost but problem with GUI and with reliable wifi is not working cannot find any AP

Tuxman2, 2017-01-17:
@Haxk20: They put: tsc=reliable force_tsc_stable=1 clocksource=tsc clocksource_failover=tsc
but I don't know if it can solve your problem. Linux Community try to find the solution but it is not easy.

Haxk20, 2017-01-22:
So im not the only one having problem with tsc ? And what clocksource do you use and what filesystem ?f2fs or ext4 ?

Haxk20, 2017-01-22:
But weird is that when i use only clocksource=tsc and switch to cli tablet is totally usable no problem but when i use gui its start to do weird stuff so i think problem is with i915 is there some patch for i915 and clocksource maybe like i915 speed or something

Haxk20, 2017-01-22:
????

Haxk20, 2017-02-02:
Ok so updated to 4.9.6-ARCH kernel and used those kernel parameters and tadaaa it worked a lot better but wifi is stuck at searching for AP (wifi) with those kernel parameters but when used only clocksource=tsc then wifi worked but not tested other thing if keyboard will do crazy will test and will post here

Haxk20, 2017-03-10:
Well yeah if only clocksource  = tsc is applied the tablet is unstable again so yeah those parameters are perfect for me from jiffies

Haxk20, 2017-03-17:
Closing because its fixed with kernel parm.

closed 2017-03-17 by Haxk20

nicman23, 2017-03-17:
@Haxk20 to be clear you use all 3 ( `tsc=reliable force_tsc_stable=1 clocksource=tsc clocksource_failover=tsc` ) ?

Haxk20, 2017-03-18:
Without clocksource=tsc but all others