burzumishi/linux-baytrail-flexx10 #35: New error message with dmesg command

Tuxman2, 2017-02-20:
Hello everybody,

well, I compiled a custom kernel (4.8.xx) with the informations given by RussianNeuroMancer. The compilation was good and my custom kernel work. But I see a lot error messages with the dmesg command. I don't know if you can help me to find a solution for each error messages. Here is the errors:

`[    0.000000] efi: requested map not found.
[    0.000000] esrt: ESRT header is not in the memory map.
`

`[    0.484358] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.484368] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.484377] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9999368bb848), AE_NOT_EXIST (20160422/psparse-542)
[    0.484441] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.484449] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.484457] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9999368bb848), AE_NOT_EXIST (20160422/psparse-542)
[    0.484472] ACPI Error: Method execution failed [\_SB.I2C2.CAM1._STA] (Node ffff9999368bb848), AE_NOT_EXIST (20160422/uteval-103)
[    0.484595] ACPI: Power Resource [P28P] (off)
[    0.484719] ACPI: Power Resource [P18P] (off)
[    0.485202] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.485211] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.485218] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9999368bbaf0), AE_NOT_EXIST (20160422/psparse-542)
[    0.485281] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.485289] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.485297] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9999368bbaf0), AE_NOT_EXIST (20160422/psparse-542)
[    0.485312] ACPI Error: Method execution failed [\_SB.I2C2.CAM2._STA] (Node ffff9999368bbaf0), AE_NOT_EXIST (20160422/uteval-103)
[    0.485714] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.485722] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.485730] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9999368bb000), AE_NOT_EXIST (20160422/psparse-542)
[    0.485792] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.485799] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.485807] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9999368bb000), AE_NOT_EXIST (20160422/psparse-542)
[    0.485822] ACPI Error: Method execution failed [\_SB.I2C2.CAM3._STA] (Node ffff9999368bb000), AE_NOT_EXIST (20160422/uteval-103)
[    0.486294] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.486303] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.486311] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9999368bb208), AE_NOT_EXIST (20160422/psparse-542)
[    0.486373] ACPI Error: No handler for Region [GPOP] (ffff9999368b6f78) [GeneralPurposeIo] (20160422/evregion-166)
[    0.486380] ACPI Error: Region GeneralPurposeIo (ID=8) has no handler (20160422/exfldio-299)
[    0.486388] ACPI Error: Method parse/execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9999368bb208), AE_NOT_EXIST (20160422/psparse-542)
[    0.486403] ACPI Error: Method execution failed [\_SB.I2C2.CAM4._STA] (Node ffff9999368bb208), AE_NOT_EXIST (20160422/uteval-103)
[    0.496907] ACPI: Power Resource [P28X] (off)
[    0.497055] ACPI: Power Resource [P18X] (off)
[    0.497196] ACPI: Power Resource [P28T] (off)
[    0.497328] ACPI: Power Resource [P18T] (off)
[    0.497456] ACPI: Power Resource [P1XT] (off)
[    0.508140] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    0.508152] acpi PNP0A08:00: _OSC: OS supports [ASPM ClockPM Segments MSI]
[    0.508278] acpi PNP0A08:00: _OSC failed (AE_ERROR); disabling ASPM
[    0.508308] acpi PNP0A08:00: fail to add MMCONFIG information, can't access extended PCI configuration space under this bridge.`

`[    2.356072] ACPI Error: No handler for Region [XSCG] (ffff9999368bdaf8) [GenericSerialBus] (20160422/evregion-166)
[    2.356082] ACPI Error: Region GenericSerialBus (ID=9) has no handler (20160422/exfldio-299)
[    2.356091] ACPI Error: Method parse/execution failed [\_SB.ADP1._PSR] (Node ffff9999368c6b18), AE_NOT_EXIST (20160422/psparse-542)
[    2.356111] ACPI Exception: AE_NOT_EXIST, Error reading AC Adapter state (20160422/ac-128)`

`[    3.013249] axp20x-i2c i2c-INT33F4:00: AXP20x variant AXP288 found
[    3.013391] axp20x-i2c i2c-INT33F4:00: Failed to get gpio for extcon pdata
[    3.013571] axp288_get_acpi_cdata: BCFG table found, size=189
[    3.013574] axp288_get_acpi_cdata: battid and fg_name mismatch!!!
[    3.013709] axp288_get_fg_config_data: Loading default fg config.
[    3.207703] usb 1-2: new high-speed USB device number 2 using xhci_hcd
[    3.216153] axp20x-i2c i2c-INT33F4:00: added platform data for axp288_extcon
[    3.216164] axp20x-i2c i2c-INT33F4:00: added platform data for axp288_charger
[    3.216173] axp20x-i2c i2c-INT33F4:00: added platform data for axp288_fuel_gauge
[    3.254053] axp288_extcon axp288_extcon: failed to get otg transceiver, continuing anyway

[    3.924204] axp288_charger axp288_charger: failed to register otg notifier`

`[    7.356382] hctosys: unable to open rtc device (rtc0)

[    7.358078] PM: Hibernation image not present or could not be loaded.

[    8.026019] i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed

[   12.870631] ACPI Warning: Could not enable fixed event - RealTimeClock (4) (20160422/evxface-654)

[   37.387723]  Baytrail Audio Port: ASoC: no backend DAIs enabled for Baytrail Audio Port`

Here is the result for Baytrail audio: 

`$ sudo dmesg | grep 5640
[   13.566296] bytcr_rt5640 bytcr_rt5640: BIOS Routing: AIF1 connected
[   13.566303] bytcr_rt5640 bytcr_rt5640: quirk IN3_MAP enabled
[   13.566306] bytcr_rt5640 bytcr_rt5640: quirk DIFF_MIC enabled
[   13.566309] bytcr_rt5640 bytcr_rt5640: quirk SSP0_AIF1 enabled
[   13.566311] bytcr_rt5640 bytcr_rt5640: quirk MCLK_EN enabled
[   13.676907] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[   13.677013] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[   13.699599] bytcr_rt5640 bytcr_rt5640: rt5640-aif1 <-> ssp0-port mapping ok`

Thanks in advance if you can help me.

RussianNeuroMancer, 2017-02-21:
I have no idea about ACPI errors.

For AXP288 errors - try another patchset (from another revision) or better wait for 4.11rc1. There should be working AXP288 driver (besides missing Whiskey Cove PMIC support).

>  I don't know if I must blacklist battery in /etc/modprobe.d/blacklist.conf.

No harm if you try that.

> About baytrail sound problem

With upstream 4.10 no sound too? You sure you put UCM in right location?

nicman23, 2017-02-22:
> With upstream 4.10 no sound too

did that intel dev upstream the detection of the right firmware / quirks, or do we still need to add a quirk table?

Tuxman2, 2017-02-22:
Hi,

@RussianNeuroMancer: I put UCM in /usr/share/alsa/  
I've got a  /usr/share/alsa/ucm//bytcr-rt5640 directory.
I will try your 4.10 custom kernel to see if I have the same issues as my personnal custom kernel.
With my custom kernel I had a kernel panic during the boot phase.

Tuxman2, 2017-02-23:
Hi,

I tested the 4.10 custom kernel (Rev2) from RussianNeuroMancer and it works without problem. I checked with sudo dmesg command and I have still some errors with his kernel:

[    0.000000] efi: requested map not found.
[    0.000000] esrt: ESRT header is not in the memory map.
[    2.988924] axp20x-i2c i2c-INT33F4:00: Failed to get gpio for extcon pdata
[    2.989026] axp288_get_acpi_cdata: battid and fg_name mismatch!!!
[    3.205827] axp288_extcon axp288_extcon: failed to get otg transceiver, continuing anyway
[    3.879577] axp288_charger axp288_charger: failed to register otg notifier
[    7.775353] i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed
and the ACPI errors indicated in the first message.

About my sound problem, I have a good news and a bad news. For the good news, now I don't have anymore the 'Baytrail Audio Port: ASoC: no backend DAIs enabled for Baytrail Audio Port' message. I downloaded the UCM-master file from plbossart website, and deleted the files I installed from jharrison022 (alsa-ucm-5640.tar.gz). 

Here is the results for sound card with dmesg:

> [   12.452713] intel_sst_acpi 80860F28:00: Detected Baytrail-CR platform
[   12.456701] intel_sst_acpi 80860F28:00: LPE base: 0x90a00000 size:0x200000
[   12.456706] intel_sst_acpi 80860F28:00: IRAM base: 0x90ac0000
[   12.456741] intel_sst_acpi 80860F28:00: DRAM base: 0x90b00000
[   12.456750] intel_sst_acpi 80860F28:00: SHIM base: 0x90b40000
[   12.456759] intel_sst_acpi 80860F28:00: Mailbox base: 0x90b44000
[   12.456766] intel_sst_acpi 80860F28:00: DDR base: 0x20000000
[   12.459648] intel_sst_acpi 80860F28:00: Got drv data max stream 25
[   12.933185] bytcr_rt5640 bytcr_rt5640: BIOS Routing: AIF1 connected
[   12.933192] bytcr_rt5640 bytcr_rt5640: quirk IN3_MAP enabled
[   12.933195] bytcr_rt5640 bytcr_rt5640: quirk DIFF_MIC enabled
[   12.933197] bytcr_rt5640 bytcr_rt5640: quirk SSP0_AIF1 enabled
[   12.933200] bytcr_rt5640 bytcr_rt5640: quirk MCLK_EN enabled
[   13.048677] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[   13.048758] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[   13.048807] compress asoc: snd-soc-dummy-dai <-> compress-cpu-dai mapping ok
[   13.071600] bytcr_rt5640 bytcr_rt5640: rt5640-aif1 <-> ssp0-port mapping ok
[   27.919513] sst-mfld-platform sst-mfld-platform: Enter: enable=1 port_name=ssp0-port
[   27.919925] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[   28.143876] intel_sst_acpi 80860F28:00: Alloc for str 3 pipe 0xe
[   33.295937] intel_sst_acpi 80860F28:00: Free for str 3 pipe 0xe
[   33.365140] sst-mfld-platform sst-mfld-platform: Enter: enable=0 port_name=ssp0-port
[   33.366342] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[   41.975506] sst-mfld-platform sst-mfld-platform: Enter: enable=1 port_name=ssp0-port
[   42.059522] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[   49.163578] intel_sst_acpi 80860F28:00: Alloc for str 3 pipe 0xe
[   51.127726] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[   76.323633] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[  155.944423] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[  170.567896] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[  216.027792] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[  221.687827] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[  239.576156] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[  239.971623] intel_sst_acpi 80860F28:00: Alloc for str 1 pipe 0x90
[  259.308809] intel_sst_acpi 80860F28:00: Free for str 1 pipe 0x90
[  261.782626] sst-mfld-platform sst-mfld-platform: Enter: enable=0 port_name=ssp0-port
[  261.785289] intel_sst_acpi 80860F28:00: Free for str 3 pipe 0xe

Here is the result in the sound applet:

> Output
Headphones Playback bytcr-rt5640  
Speaker playback bytcr-rt5640  
Monospeaker playback bytcr-rt5640

> Input
DigitalMics capture, NOT TESTED bytcr-rt5640
Headset Mic capture bytcr-rt5640
Internal IN1 analog Mic capture bytcr-rt5640
Internal IN3 analog Mic capture bytcr-rt5640

Others results:

[... 14 lines cut ...]
[   13.048677] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[   13.048758] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[   13.071600] bytcr_rt5640 bytcr_rt5640: rt5640-aif1 <-> ssp0-port mapping ok

> sudo dmesg | grep bytcr_rt5640
[   12.933185] bytcr_rt5640 bytcr_rt5640: BIOS Routing: AIF1 connected
[   12.933192] bytcr_rt5640 bytcr_rt5640: quirk IN3_MAP enabled
[   12.933195] bytcr_rt5640 bytcr_rt5640: quirk DIFF_MIC enabled
[   12.933197] bytcr_rt5640 bytcr_rt5640: quirk SSP0_AIF1 enabled
[   12.933200] bytcr_rt5640 bytcr_rt5640: quirk MCLK_EN enabled
[   13.048677] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[   13.048758] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[   13.071600] bytcr_rt5640 bytcr_rt5640: rt5640-aif1 <-> ssp0-port mapping ok

> cat /proc/asound/pcm
00-00: 1 :  : playback 1 : capture 1
00-01: Deep-Buffer Audio (*) :  : playback 1

> cat /proc/asound/cards
 0 [bytcrrt5640    ]: bytcr-rt5640 - bytcr-rt5640
                      bytcr-rt5640

I have the asound.state file downloaded from here: http://home.ix.netcom.com/%7Ejjingber/asound.state in the /var/lib/alsa directory.

The bad news is that I haven't still sound whether through the headphones, the speakers, or via mono speakers. And when I perform a sound test via the sound applet, I get no sound. So I don't know what is the problem.

RussianNeuroMancer, 2017-02-25:
@Tuxman2 
My build based on 4.8.10 sources, not 4.10. Possible reason for kernel panic of your own build could be that some patches not applied properly. Pay attention for "patch" utility output - some patches should be applied manually if automatic patching failed. Also you need to patch exactly 4.8.10 source code (not for example 4.8.14) and maybe use same build options (from Readme.txt). 
Question about OTG should be addressed to patches authors, not to me. Also number of revision doesn't mean that rev2 is newer or better than rev1, it's just another patchset, another approach. rev1 could be better for you.
You doesn't need anyone's asound.state, try to remove it, reboot and check again. Also test audio in upstream Linux 4.10, 4.10-fixes branch https://github.com/plbossart/sound/tree/topic/v4.10-fixes or 4.11rc1, or drm-intel-next-queued-cht branch https://github.com/jwrdegoede/linux-sunxi/commits/drm-intel-next-queued-cht as soon as it get rebased  on 4.11 (currently based on 4.10rc5).

@nicman23 unfortunately I don't remember about audio output, probably should be detected automatically. For mic quirks still required, for example 4.11rc1 will get quirk that theoretically should be valid for all Insyde-devices (with Insyde DMI string).

Tuxman2, 2017-02-26:
@RussianNeuroMancer: Well, I now understand why I had a problem with my custom kernel. I applied all your patches with a 4.8.17 source kernel. Ok, I must use the 4.8.10 kernel source. Until the 4.11-rc kernel will be available, I will use the 4.8.10 kernel. To solve all the errors from dmesg that I have (see my first message) I need to know what other patches/fixes should I apply. Perhaps people and you here can tell me where can I find all the necessary patches to solve it ?

Thanks.

RussianNeuroMancer, 2017-02-26:
> To solve all the errors from dmesg that I have (see my first message) I need to know what other patches/fixes should I apply. Perhaps people and you here can tell me where can I find all the necessary patches to solve it ?

As I understand AXP288 and audio errors should be gone eventually (maybe with upstream 4.11 or maybe with drm-intel-next-queued-cht  branch) others are not kernel issues (hibernation and rt) and ACPI errors should be reported to hardware vendor (it's not Linux fault that it detect and tells you about errors in board firmware).

Tuxman2, 2017-02-27:
@RussianNeuroMancer: I talked with people like Hans, and these people told me to use a recent kernel (example : the last 4.10.1 stable kernel). Use a recent kernel can solve some of my errors messages. Hans also talked to me about his drm-intel-next-queued-cht branch with patches who can solve some of my errors messages (example: efi: requested map not found). Yes, it's a good idea, but as I said to him, it's no a good idea to apply any patches without advice. Some are experimental patches that can break the hardware. You gave all the informations to make a custom kernel with 4.8.10 version and I thank you. I know you want to wait 4.11 kernel to give us all the necessary informations to make a new custom kernel, but as the 4.11-rc1 kernel is not available and as we don't know when it will be released, It could be a good idea to tell us (if you know or others people here) what are the necessary patches to apply on the 4.10 series of the kernel. Can we apply the patches from the burzumishi github (4.8) ? What are the necessary patches for baytrail sound, for battery status, ...., on the 4.10.x kernel ? As I said to Hans, without advices we are lost.

RussianNeuroMancer, 2017-02-27:
>  I know you want to wait 4.11 kernel to give us all the necessary informations to make a new custom kernel

Actually I want get 4.11 working just to be able to upgrade couple of my BayTrail devices to Ubuntu Gnome 17.04. Information about utilized patches is side effect.

> but as the 4.11-rc1 kernel is not available and as we don't know when it will be released

It's not like it will take more than week or two.

> if you know

I doesn't know. I wasn't able to get even 4.8.14 patched and working (some patch probably need to be rewritten, but I don't clearly remember which one). That why I waiting 4.11 - to deal with less patches in first place (4.10 will need more patches than 4.11).

> Can we apply the patches from the burzumishi github (4.8) ? 

This patches is listed in Readme.txt.

> What are the necessary patches for baytrail sound, for battery status, ...., on the 4.10.x kernel ? 

There is easy answer for audio: https://github.com/plbossart/sound/commits/topic/v4.10-fixes (latest six commits as I understand).

I doesn't have answer about anything else.

Tuxman2, 2017-02-27:
@RussianNeuroMancer:  I downloaded the 4.10.1 kernel from the kernel mainline PPA Ubuntu website. I will install the compiled deb packages made by them, then I could see what works and what does not work on my hardware. I will try to contact plbossart to know what exactly patches we must apply on the 4.10.x kernel. I will ask Hans to know what exactly patches we must apply for battery status for 4.10.x kernel. If I have all the necessary informations, I will post it here. ;-)

Tuxman2, 2017-02-28:
Well, I tested the 4.10.1 kernel from Kernel Mainline PPA Ubuntu (without patches). I have less errors with dmesg command than before, and the sound card is well initialized (with the UCM files), but the 'Atom PMC platform clocks' patch is always needed. I don't know if I always need the burzumishi patches and the 4.8fix-patchset from RussianNeuroMancer. I will try to compile a 4.10.1 custom kernel when I will have all informations about patches to apply for the 4.10.x version.

RussianNeuroMancer, 2017-02-28:
> 4.8fix-patchset from RussianNeuroMancer

It's was from https://github.com/plbossart/sound/commits/topic/v4.8-fixes
Now use https://github.com/plbossart/sound/commits/topic/v4.10-fixes instead.

Tuxman2, 2017-02-28:
@RussianNeuroMancer: I try to have all the informations for the 4.10.x kernel (what are the necessary patches for this kernel).

RussianNeuroMancer, 2017-02-28:
https://bugzilla.kernel.org/show_bug.cgi?id=109051#c743

Tuxman2, 2017-03-01:
@RussianNeuroMancer: For cstate problem I will look it later. ;-)

Well, I have informations for Baytrail sound card rt5640 support on 4.10.x kernel. For those who are interested, here are the patches you can try on the 4.10.1 kernel. Patches from https://github.com/plbossart/sound/commits/topic/v4.10-fixes

> ASoC: Intel: boards: remove .pm_ops in all Atom/DPCM machine drivers
ASoC: Intel: atom: fix frame polarity
ASoC: rt5640: Add "10EC3276" ACPI ID
clk: Make x86/ conditional on CONFIG_COMMON_CLK
clk: x86: Add Atom PMC platform clocks
arch/x86/platform/atom: Move pmc_atom to drivers/platform/x86
platform/x86: Enable Atom PMC platform clocks

Additional patches:

> platform/x86:fix typo in comment
ASoC: core: Add API to use DMI name in sound card long name

I tried to contact Hans de Goede to have informations about support for battery for 4.10.x kernel. But I have not yet reply from him. Then I don't know what we must apply as patches for axp288 and axp20 on the 4.10.x kernel. :-( 
I don't know if we must applied others patches for 4.10.x kernel as the drm-intel-next-queued-cht branch have many patches for all problems (acpi, input, i2c, etc.). I can solve my 'efi: requested map not found' and 'ESRT header is not in the memory map' message errors, but for my ACPI errors problem, I don't know if I must apply these patches: ACPICA: Log Exceptions and Errors as warning while loading extra tables and ACPICA: Detect duplicate SSDT tables ? If you have an idea I joined my dmesg file (4.10.1 kernel without patches included) to this topic.

Finaly, I don't know if the burzumishi patches are always necessary for 4.9.x kernel and above as they were made for 4.8.xx kernel ?

[kernel4101ubuntuppa.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/810888/kernel4101ubuntuppa.txt)

Tuxman2, 2017-03-06:
Well, The 4.11-rc1 kernel is out, I will test it on my hardware to see if there are improvements. I downloaded the 4.11-rc1 sources if I need to do some changes. I will tell you if there is good and bad news after doing the test.

RussianNeuroMancer, 2017-03-06:
@Tuxman2 Please let me know if you find a way to build rtl8723bs driver. It's not building with 4.11rc1 for me.

Tuxman2, 2017-03-07:
Well, I tested compiled packages from Ubuntu kernel Mainline PPA for the 4.11-rc1 kernel. I have good news. :-)

First, the battery seems to be supported 'out-of-the-box'. As the axp20 and axp288 options are defined as modules in the compiled version from Ubuntu kernel Mainline PPA, I needed to launch axp20 and axp288 modules in /etc/modules-load.d/axp288.conf like someone said here. I will try to compile a custom 4.11-rc1 kernel with the axp20 and axp288 options to yes. In my test, the device was connected to the AC adapter, and it showed me that it was charging. The only problem that I saw is that it shows me 2 batteries in the taskbar applet for battery (Intel SR 1 SR real battery 100% and Secondary battery).

Second, the baytrail rt5640 sound card seems to be supported 'out-of-the-box'. The platform clock patch seems to be available by default in this kernel, because I don't have this dmesg error message anymore: "bytcr_rt5640 bytcr_rt5640: Failed to get MCLK from pmc_plt_clk_3: -2". In my case, I still have no sound but it seems to be a problem with the pulseaudio program because all things are well initialized.

Last, in my case, I always have the following dmesg errors (acpi  error: No handler for Region [GPOP], efi: requested map not found,  i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed, ...). I don't know if there are patches for that, but I posted these bugs on the bugzilla.kernel.org website.

For more details, see my dmesg file here:

[dmesg411rc1.txt](https://github.com/burzumishi/linux-baytrail-flexx10/files/824296/dmesg411rc1.txt)

Note:  Do you think if these patches are always necessary ?:

0001-mmc_sdhci_on_intel_baytrail.patch
0006-mmc_disable_crc_checks.patch
0008-rtl8723bs_rfill_bluetooth.patch
0009-intel_i915_on_baytrail.patch
0010-intel_i915_on_baytrail.patch
0011-intel_i915_on_baytrail.patch
0012-intel_i8042_disable.patch
0015-acpi_statically_assign_irq_in_reduced_mode.patch
0016-intel_i915_use_legacy_turbo.patch
0017-intel_i915_avoid_agressive_downclocking_on_baytrail.patch

RussianNeuroMancer, 2017-03-07:
> Note: Do you think if these patches are always necessary ?:

I didn't review what patches is still required for 4.11. mmc stuff seems like still needed for stability with WiFi, but I not seen version of this patches that adjusted for 4.11. rtl8723bs rfill looks like needed too. rtc by default prevent volume buttons from working, so something need to be done about rtc too.

What I am missing, besides very device-specific patches? (audio, touchscreens)

Tuxman2, 2017-03-07:
@RussianNeuroMancer: I have problem with the buttons for audio volume (buttons on the side of the tablet), it still doesn't work for me (I have an error message in dmesg).  I think I must apply the patch that people gave here.

RussianNeuroMancer, 2017-03-07:
I mentioned that:

> rtc by default prevent volume buttons from working, so something need to be done about rtc too.

Anything else?

Tuxman2, 2017-03-07:
I think that you said somewhere else that if we enable some options in the kernel, it enable the volume buttons.

RussianNeuroMancer, 2017-03-07:
I will try to find out what need to be done to make volume buttons works again with 4.11.

Tuxman2, 2017-03-08:
@RussianNeuroMancer: I've got a patch for volume buttons problem. I will test it on 4.11-rc1 kernel. It's only one line to modify into the arch/x86/kernel/acpi/boot.c file in the kernel sources.
I guess that you tested the 4.11-rc1 kernel with your hardware ? How does it work ?

RussianNeuroMancer, 2017-03-09:
> I've got a patch for volume buttons problem. I will test it on 4.11-rc1 kernel. It's only one line to modify into the arch/x86/kernel/acpi/boot.c file in the kernel sources.

You mean [this one](https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/kernel/patches/v4.8/0015-acpi_statically_assign_irq_in_reduced_mode.patch)?

Please provide info about your device here: https://bugzilla.kernel.org/show_bug.cgi?id=115271

> I guess that you tested the 4.11-rc1 kernel with your hardware ? How does it work ?

I waiting for updating of Hans tree, then I will try to build kernel with additional fixes for buttons, [phantom battery](https://bugzilla.kernel.org/show_bug.cgi?id=88471#c100), [latest antifreeze](https://bugzilla.kernel.org/show_bug.cgi?id=109051#c742), [rfkill patch](https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/kernel/patches/v4.8/0008-rtl8723bs_rfill_bluetooth.patch), [disabling ps/2 controller patch](https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/kernel/patches/v4.8/0012-intel_i8042_disable.patch). I also will need audio patch for DEXP Ursus 10XW.

Tuxman2, 2017-03-09:
@RussianNeuroMancer: 

> You mean this one?

Yes.

> Please provide info about your device here: https://bugzilla.kernel.org/show_bug.cgi?id=115271

I made a lot of bugs reports on the kernel bugzilla website about all my dmesg error messages

> I waiting for updating of Hans tree, then I will try to build kernel with additional fixes for buttons, phantom battery, latest antifreeze, rfkill patch, disabling ps/2 controller patch. I also will need audio patch for DEXP Ursus 10XW.

Ok. Could you tell me if others patches are necessary for the 4.11-rc1 kernel when you will receive all the informations ? For this bug: https://bugzilla.kernel.org/show_bug.cgi?id=194685 Hans de Goede have 2 patches that I can apply.

I have an issue with the baytrail sound (no sound). It seems that I must use SSP0_AIF2 rather than SSP0_AIF1 but I must modify the sound/soc/intel/boards/bytcr_rt5640.c file in the kernel source to ignore the BIOS routing to select SSP0_AIF2 (I guess by default). But I don't know how to do that. Before that the UCM files must be set-up, and I need a user-space config on top of the kernel. What does it means ?

Thanks.

RussianNeuroMancer, 2017-03-09:
> Could you tell me if others patches are necessary for the 4.11-rc1 kernel when you will receive all the informations ? 

Listed patches seems like is enough, and some probably even excessive, like ps/2 (@nicman23 is there any particular reason for disabling ps/2 on BayTrail?) One missing piece that make me worry is lack of SDHCI patch that seems like missing for 4.11.  Could you please check if 4.11 can be built and works with [this patch](https://github.com/hadess/rtl8723bs/issues/76#issuecomment-272932161)?

> It seems that I must use SSP0_AIF2 rather than SSP0_AIF1

You sure you need AIF2 instead of SSP2 for example? How do you find out that you need AIF2?

> But I don't know how to do that.

Is your device is listed [here](https://github.com/torvalds/linux/blob/master/sound/soc/intel/boards/bytcr_rt5640.c#L318)?

Tuxman2, 2017-03-09:
> Listed patches seems like is enough, and some probably even excessive, like ps/2 (@nicman23 is there any particular reason for disabling ps/2 on BayTrail?) One missing piece that make me worry is [...]

I will made a custom kernel from 4.11-rc1 this week, I will test the patches you said above (without  disabling ps/2 controller patch), and others patches for my hardware. I also will try the patch you given here.

> You sure you need AIF2 instead of SSP2 for example? How do you find out that you need AIF2?

I talked to plbossart about my sound problem. He said to me to use SSP0_AIF2 rather than SSP0_AIF1 because SSP0_AIF2 seems to work better for different platforms.

> Is your device is listed here?

No my device is not listed here. :-(

RussianNeuroMancer, 2017-03-09:
> No my device is not listed here. :-(

So you need to add it then. Did you published dmidecode output anywhere?

Tuxman2, 2017-03-09:
Yes I gave it to plbossart, but he don't use the informations for quirk.

RussianNeuroMancer, 2017-03-09:
Try to add quirk by using this as reference

```
	{
		.callback = byt_rt5640_quirk_cb,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "MEDION"),
			DMI_MATCH(DMI_PRODUCT_NAME, "Akoya E2212T MD99720"),
		},
		.driver_data = (unsigned long *)(BYT_RT5640_SSP0_AIF2),

	},
```

However, I put only BYT_RT5640_SSP0_AIF2 here, and you probably need some other quirks for your device. Ass you can see [here](https://github.com/torvalds/linux/blob/16b5711485b0eba250451b3a85ab869939f20c7d/sound/soc/intel/boards/bytcr_rt5640.c#L318), there is many other variants. Ask Pierre what else you could need and add other quirks into driver_data, like it was done with other devices.

Tuxman2, 2017-03-09:
@RussianNeuroMancer: Ok, thanks. ;-) 

I will ask Pierre if it is no more easy for me to add the quirk for my device than to modify the bytcr_rt5640.c file in the kernel sources.

Tuxman2, 2017-03-10:
Well, I asked Pierre and I need to find myself the right options (quirk table for sound) for my device.

I can try these 2 configurations:

.driver_data = (unsigned long *)(BYT_RT5640_IN1_MAP |
						   BYT_RT5640_MCLK_EN |
                                                   BYT_RT5640_SSP0_AIF2),
or

.driver_data = (unsigned long *)(BYT_RT5640_IN1_MAP |
                                                   BYT_RT5640_MONO_SPEAKER |
                                                   BYT_RT5640_DIFF_MIC |
                                                   BYT_RT5640_SSP0_AIF2 |
                                                   BYT_RT5640_MCLK_EN
                                                   ),

RussianNeuroMancer, 2017-03-11:
Patch that unregister second battery: https://bugzilla.kernel.org/show_bug.cgi?id=194811#c3

Can you test this, please?

Tuxman2, 2017-03-12:
@RussianNeuroMancer: Ok. I downloaded them. I was ready to compile my custom kernel version but with your new patches I have to start over at the beginning.
I want to tell to you that the patches given by you above are well taken into account without errors with the 'patch -p1 < ....' command (latest antifreeze, ...)

Tuxman2, 2017-03-13:
@RussianNeuroMancer: I used the debian/rules commands to make my custom kernel from 4.11-rc1 with patches. I have a question for you. Well, in an other topic here, you told me to copy the files from debian.master directory to debian directory and to rename debian/control.stub.in to debian/control. My question is: must I copy all the debian.master directory to debian directory, or must I copy the files without directory from debian.master to debian directory ? In my case, I copy all the debian.master directory (files + directories) to debian directory, and I left the debian.master directory as it was.

I will test my custom kernel (with the patches you given to me) and I will tell you later if it works or not.

RussianNeuroMancer, 2017-03-13:
Content (folders and files) of debian.master into debian.

Tuxman2, 2017-03-14:
> Content (folders and files) of debian.master into debian.
The debian.master folder can be deleted or not after that ?

Well, I tested my custom kernel and it works. I have good news.

What works: 
- Patches that unregister second battery.
- Latest antifreeze patch (I remove intel_idle.max_cstate=0 and intel_pstates=disabled lines in GRUB. I didn't see freeze but I didn't use the hardware during a long time).
- Volume buttons patch.
- Battery support seems to work 'out-of-the-box'.
- Bluetooth.
- For wifi, I applied the patches given by RussianNeuroMancer, but I didn't try wifi.

Note: For the battery, Here are the options I defined in the kernel:

CONFIG_AXP20X_POWER=y
CONFIG_AXP288_ADC=y
CONFIG_AXP288_CHARGER=y
CONFIG_AXP288_FUEL_GAUGE=y
CONFIG_INPUT_AXP20X_PEK=y
CONFIG_EXTCON_AXP288=y
CONFIG_EXTCON=y
CONFIG_MFD_AXP20X=y
CONFIG_MFD_AXP20X_I2C=y
CONFIG_REGULATOR_AXP20X=y

What don't work:
- Sound. I have still not sound yet

Note: I made a quirk entry:
{
  .callback = byt_rt5640_quirk_cb,
  .matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "MEDION"),
			DMI_MATCH(DMI_PRODUCT_NAME, "Akoya E2212T MD99720"),
		    },
   .driver_data = (unsigned long *)(BYT_RT5640_IN1_MAP |
						     BYT_RT5640_MCLK_EN |
						     BYT_RT5640_SSP0_AIF2)),
},

Here is the result with dmesg : 

[    9.508994] bytcr_rt5640 bytcr_rt5640: BIOS Routing: AIF1 connected
[    9.509002] bytcr_rt5640 bytcr_rt5640: quirk IN1_MAP enabled
[    9.509005] bytcr_rt5640 bytcr_rt5640: quirk SSP0_AIF2 enabled
[    9.509008] bytcr_rt5640 bytcr_rt5640: quirk MCLK_EN enabled
[    9.524720] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> media-cpu-dai mapping ok
[    9.525030] bytcr_rt5640 bytcr_rt5640: snd-soc-dummy-dai <-> deepbuffer-cpu-dai mapping ok
[    9.525076] compress asoc: snd-soc-dummy-dai <-> compress-cpu-dai mapping ok
[    9.526270] bytcr_rt5640 bytcr_rt5640: rt5640-aif2 <-> ssp0-port mapping ok

With the pulseaudio -vvv command, I've got the pa_pid_file_create() failed error.

Now I have to find the solution to make the sound works.

RussianNeuroMancer, 2017-03-15:
> The debian.master folder can be deleted or not after that ?

I didn't tried deleting it.

> For the battery, Here are the options I defined in the kernel

Should work without changing config.

> I have still not sound yet

You 100% sure that UCM-files in right location? If yes, try [second quirk](https://github.com/burzumishi/linux-baytrail-flexx10/issues/35#issuecomment-285640209) when you will make rc2 build, and also could you please add this quirk and publish result deb-files somewhere? 

> 	{
> 		.callback = byt_rt5640_quirk_cb,
> 		.matches = {
> 			DMI_MATCH(DMI_SYS_VENDOR, "DEXP"),
> 		},
> 		.driver_data = (unsigned long *)(BYT_RT5640_IN3_MAP |
> 						 BYT_RT5640_MCLK_EN |
> 						 BYT_RT5640_SSP0_AIF1),
> 
> 	},

Tuxman2, 2017-03-15:
> The debian.master folder can be deleted or not after that ?
> I didn't tried deleting it.

Ok. As I said before I copied all files (files + directories) from debian.master directory to Debian directory, and left the debian.master directory as it was.

> For the battery, Here are the options I defined in the kernel
> Should work without changing config.

Some options for the battery are defined as modules by default in the kernel. Then it is necessary to load these modules at boot with a configuration file. As I defined yes to these options,  I don't need a configuration file (example: axp288.conf). the CONFIG_REGULATOR_AXP20X option is necessary or not ?

> You 100% sure that UCM-files in right location? If yes, try second quirk when you will make rc2 build

Yes. The location for UCM files is: /usr/share/alsa/ucm
I put the bytcr-rt5640 directory from UCM files to it.

> and also could you please add this quirk and publish result deb-files somewhere?
OK. If I have time I will do it. Is it possible to post the deb files here ?

RussianNeuroMancer, 2017-03-15:
> Is it possible to post the deb files here ?

You probably will need to upload it somewhere.

Tuxman2, 2017-03-16:
@RussianNeuroMancer:  I will make a custom kernel from the 4.11-rc2 kernel for your hardware (with the patches and your quirk included). I will try to give you the deb packages from here (github) as soon as possible. I have to compile a custom kernel for myself too.

nicman23, 2017-03-17:
@Tuxman2 if you want i have an nextcloud server on https://snf-738721.vm.okeanos.grnet.gr/ . You can upload there with the usr/pasw of guest.

jusss, 2017-03-17:
@RussianNeuroMancer I tried 4.11-rc2, acpi works,  I compiled the kernel and got 12GB data, I am not good at config kernel, can you pass me your .config file for kernel? thanks

Tuxman2, 2017-03-17:
@nicman23: Thanks to you. I used your nextcloud server to give the deb files to RussianNeuroMancer. ;-)

@RussianNeuroMancer: You will find my deb packages for your hardware here: https://snf-738721.vm.okeanos.grnet.gr/index.php/s/tdBkXZe5JiwReHj  (411-rc2CustomKernel).
Those deb packages  contain the patches you gave to me, as well as the quirk table patch for the audio hardware of your device. All the options for the battery are set to yes by default (you will have the full support for battery), I also enabled otg support and the cpu governor is set to conservative by default.

Could you tell me if everything is ok after testing ?

RussianNeuroMancer, 2017-03-17:
@Tuxman2 

> You will find my deb packages for your hardware here

Thanks! I will check as soon as I get DEXP Ursus 10XW from service.

> Could you tell me if everything is ok after testing ?

Sure.

> I also enabled otg support 

What this give when regular Ubuntu running on hardware? (Not Android.)

> and the cpu governor is set to conservative by default

btw, check Schedutil governor.

Tuxman2, 2017-03-17:
@RussianNeuroMancer: You're welcome. ;-)

> I also enabled otg support
> What this give when regular Ubuntu running on hardware? (Not Android.)

Well, it enables the otg usb support. I've got it on my Intel hybrid tablet, and I've got an adapter for micro-usb to usb. I can use an usb key with it but for now I haven't got test it yet.

Hurray !!! I've got sound now. :-D 
I put all the files and directories from UCM-master.zip (from plbossart website) in /usr/share/alsa/ucm. I defined a quirk patch for my sound hardware in /sound/soc/intel/boards/bytcr_rt5640.c before compiling my custom kernel. Here is what I defined for .driver_data part:

`.driver_data`` = (unsigned long *)(BYT_RT5640_IN1_MAP |
						 BYT_RT5640_MCLK_EN |
						 BYT_RT5640_SSP0_AIF1),

The headphones, the speakers, the mono-speakers work without problems. But I don't know if I must define IN1_MAP or IN3_MAP. I need to test the microphone. It will be done this weekend. I will send to plbossart the quirk patch for my hardware when it will be ok.

Else, with v4.11-rc2, I always got these dmesg errors messages (see above)  :

ACPI Error: No handler for Region [GPOP]
....
....
ACPI Error: No handler for Region [XSCG]
i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed
ACPI Warning: Could not enable fixed event - RealTimeClock (4) 

I hope that the kernel team will find the solutions in the next 4.11RC versions.

nicman23, 2017-03-17:
yeah basically the last big thing we are missing is a clock...

RussianNeuroMancer, 2017-03-18:
@Tuxman2  I thought enabling OTG support is about using tablet is "client" device for PC (like Android-devices that communicate via MTP). So, seems like this is other way around, about attaching other devices to tablet, right? But for me it's works without without any changes, not matter what dmesg errors claim about OTG. In your case you can't attach flash drive without enabling OTG support?

@nicman23 Clock patch missing in 4.10, but isn't it was upstreamed in 4.11?

Tuxman2, 2017-03-19:
>  In your case you can't attach flash drive without enabling OTG support?

I haven't yet tested this kind of thing because I was trying to operate the battery management and the sound of the audio card since a while.

With latest antifreeze patch, could you tell me if this line is always necessary or not in GRUB_CMDLINE_LINUX ?: tsc=reliable force_tsc_stable=1 clocksource=tsc clocksource_failover=tsc 
Could you also tell me if the latest antifreeze patch solved the c-state problem definitively ?

I also want to know if FTSC1000 touchscreen can be assimilated with Goodix driver ?

Thanks.

RussianNeuroMancer, 2017-03-19:
> could you tell me if this line is always necessary or not 

If you remove this line and refined jiffies never get selected as default clock source (check dmesg) then this line is not necessary. However, if refined jiffies is get selected as default clock source even sometimes, you better keep this line for now.

> Could you also tell me if the latest antifreeze patch solved the c-state problem definitively ?

C-State freeze should be solved, however there is no sdhci antifreeze patch, right? Then freezes from WiFi will happen from time to time. I doesn't get DEXP Ursus 10XW on hands yet, probably on next week.

> I also want to know if FTSC1000 touchscreen can be assimilated with Goodix driver ?

@nicman23 what situation with this touchscreen?

Tuxman2, 2017-03-19:
@RussianNeuroMancer: Here are the informations from my dmesg command :

[    0.000000] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 30582079743023 ns
[    0.558649] clocksource: Switched to clocksource refined-jiffies

As you can see in my case, this line is always needed.

@nicman23: If you have more informations with the FTSC1000 touchscreen support, tell us. ;-)
I don't know why I've got this error message through dmesg command : i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed

nicman23, 2017-03-19:
i do not have a flexx10. for me touchscreen works out of the box :/ sorry.

about wifi, it works but crashes kernel even with patches. if it doesn't there is an upstream bug in wpa_supplicant and network manager.

if you run it manually through cli it connect and you can run dhcpdc after.

RussianNeuroMancer, 2017-03-20:
@Tuxman2 

I get DEXP Ursus 10XW on hands today, but turns out service doesn't fix hardware issue so I have no choice but send it back to them under warranty. Unfortunately it will take another month until I see this tablet again. So, I can not test your packages now.

Tuxman2, 2017-03-20:
@RussianNeuroMancer: It seems to me that you had a DEXP Ursus 7 tablet ? No ?

The 4.11-rc3 kernel is out but I haven't seen any significant changes in that new version for our devices (I saw the changelog).

jusss, 2017-03-20:
.config and patches please, I couldn't wait to try it on my tablet!   :-)

RussianNeuroMancer, 2017-03-20:
Antifreeze seems like [upstreamed](It seems to me that you had a DEXP Ursus 7 tablet ? No ?).

> It seems to me that you had a DEXP Ursus 7 tablet ? No ?

Both. Ursus 7W audio goes under "Insyde" quirk that already [upstreamed](https://github.com/torvalds/linux/commit/16b5711485b0eba250451b3a85ab869939f20c7d#diff-318e6edc85450843153004e7f90f7080). Ursus 10XW need [additional quirk](https://github.com/burzumishi/linux-baytrail-flexx10/issues/35#issuecomment-286645825), that I can not test yet (last time I tested it with 4.8 it works, but need to recheck with 4.11 to send it to Pierre).

@jusss all links in previous messages, nothing is hidden :)

Tuxman2, 2017-03-21:
@RussianNeuroMancer: For i915 antifreeze it is a good news. :-)

I think that IN1_MAP doesn't work for me (nothing happens with my microphone), I will test with another custom kernel I made where I defined IN3_MAP. 
I think I will use alsa rather than pulseaudio because with pulseaudio program the sound is weaker. I must increase the volume of the sound to 75%.

RussianNeuroMancer, 2017-03-22:
@Tuxman2 
Onboard is best AFAIK.

Linux 4.11rc3 build with patches: https://yadi.sk/d/pW1Fbl-VyVPoR

Tuxman2, 2017-03-22:
@RussianNeuroMancer: Thanks. 

I close this topic and I go to the Documentation updates topic.

closed 2017-03-22 by Tuxman2