burzumishi/linux-baytrail-flexx10 #37: Rotation

kousu, 2017-08-24:
Hiya, thanks again for all your hardwork on these baytrail devices. It helps a lot!

I'm trying to get [rotation](https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/udev/rotation/lib/udev/flexx101-screen-rotate.sh) working and while the screen can rotate the image just find with `xrandr`, making the touchscreen then follow it doesn't seem to be working.

```
xinput set-int-prop $TOUCHSCREEN "Evdev Axis Inversion" 8 $INV_TO
xinput set-int-prop $TOUCHSCREEN "Evdev Axes Swap" 8 $SWAP_TO 
```

cause (in `xinput watch-props $TOUCHSCREEN` run in a second window):

```
Property 'Evdev Axis Inversion' changed.
	Evdev Axis Inversion (489):	1, 0
Property 'Evdev Axes Swap' changed.
	Evdev Axes Swap (490):	1
Property 'Evdev Axis Inversion' changed.
	Evdev Axis Inversion (489):	0, 0
Property 'Evdev Axes Swap' changed.
	Evdev Axes Swap (490):	0
```

but the touchscreen pointer doesn't follow. It still thinks that down on the screen means move the mouse down, which means the pointer doesn't follow my finger.

I don't know where to start debugging this. `dmesg` doesn't say anything useful.

kousu, 2017-08-24:
https://gist.github.com/rubo77/daa262e0229f6e398766 suggests using a coordinate transformation matrix. I'll try that next.

kousu, 2017-08-24:
the matrix method works for me.

I have a MDG Flex, model MF2IN132G.

dmidecode:

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
34 structures occupying 1952 bytes.
Table at 0x7BC18000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: INSYDE Corp.
	Version: MDG1106.1.08.020
	Release Date: 04/25/2016
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 640 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		ATAPI Zip drive boot is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 8.78
	Firmware Revision: 0.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: MDG
	Product Name: MF2IN132G
	Version: Type1 - TBD by OEM
	Serial Number: MF16062330
	UUID: 12345678-1234-5678-90AB-CDDEEFAABBCC
	Wake-up Type: Power Switch
	SKU Number: MDGM1106BFP
	Family: M1106BFP T10220310

Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
	Manufacturer: MDG
	Product Name: M1106BFP
	Version: 1.2
	Serial Number: Type2 - Board Serial Number
	Asset Tag: Type2 - Board Asset Tag
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Type2 - Board Chassis Location
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 24 bytes
Chassis Information
[... 1267 lines cut ...]
	Reporting 1 classes:
		Class originated from: 11. Type: XIKeyClass
		Keycodes supported: 248

    ↳ gpio-keys                               	id=12	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 12. Type: XIKeyClass
		Keycodes supported: 248

    ↳ Video Bus                               	id=6	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 6. Type: XIKeyClass
		Keycodes supported: 248

    ↳ SINO WEALTH USB KEYBOARD                	id=8	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 8. Type: XIKeyClass
		Keycodes supported: 248

    ↳ SINO WEALTH USB KEYBOARD                	id=13	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 13. Type: XIKeyClass
		Keycodes supported: 248

```

Tuxman2, 2017-08-25:
Hi,

have a look on my message: https://github.com/burzumishi/linux-baytrail-flexx10/issues/12#issuecomment-303996242

nicman23, 2017-09-07:
@Tuxman2 bit of an improvement, without need of inotify or yad (because i do not care about a ui for locking the rotation):

```
caser() {
  while true
    do case $1 in
    '') break ;;
    normal) xrandr --output $screen --rotate normal
      xinput set-prop "$TOUCHSCREEN" "$TRANSFORM" 1 0 0 0 1 0 0 0 1; shift 1;;
    left-up) xrandr --output $screen --rotate left
      xinput set-prop "$TOUCHSCREEN" "$TRANSFORM" 0 -1 1 1 0 0 0 0 1; shift 1;;
    bottom-up) xrandr --output $screen --rotate inverted
      xinput set-prop "$TOUCHSCREEN" "$TRANSFORM" -1 0 1 0 -1 1 0 0 1; shift 1;;
    right-up) xrandr --output $screen --rotate right
      xinput set-prop "$TOUCHSCREEN" "$TRANSFORM" 0 1 0 -1 0 1 0 0 1; shift 1;;
    *) shift 1 ;;
    esac
  done
}

main() {
  monitor-sensor|\
  while read -r line
    do [ -e $lockfile ] || caser $line
  done
}

toggle-lock() {
  if [ -e $lockfile ]
    then rm $lockfile
    else touch $lockfile
  fi
  exit
}

[ "$@" ] && [ "$@" == 'toggle' ] && toggle-lock
main
```

Tuxman2, 2017-09-08:
@nicman23: Can you make a new script with your modifications so I could test it ?

Thanks.

nicman23, 2017-09-08:
that is the script, the only thing you need to add is your TOUCHSCREEN and TRANSFORM variables from the scripted you previously posted in #12 (because i have a slightly different touchscreen from you)

to toggle rotation lock on and off just run the script with the toggle argument after its name

Tuxman2, 2017-09-10:
@nicman23: Thanks. I found another script that don't use inotify and yad. See it: https://github.com/scresante/linux-tablet-mode

Tuxman2, 2017-09-25:
@nicman23: I changed my script to looks like yours. Here is my new script: 
[rotation.sh.zip](https://github.com/burzumishi/linux-baytrail-flexx10/files/1329241/rotation.sh.zip)

I run the script on start up (after login screen) like this: sh /pathtomyhome/scripts/rotation.sh toogle (but nothing happens when I move the screen).
When I try it on the terminal screen I've got an error message : ./rotation.sh:30: .rotation.sh: Syntax error : Bad function name. When I try with ./rotation.sh toogle command only, it works but my screen rotation don't change.

Have got an idea ?  

Thanks.

nicman23, 2017-09-25:
change all toggle-lock with togglelock, you should be fine.

Tuxman2, 2017-10-03:
@nicman23: Togglelock is solved but my script still doesn't work. In the terminal, I've got that error:

> rotation.sh: 38: [: toggle: unexpected operator

The 38 line looks like it: [ "$@" ] && [ "$@" == 'toggle' ] && togglelock

nicman23, 2018-01-27:
so, did you figure it out?

Tuxman2, 2018-01-28:
@nicman23: I didn't retry your tip. I tried the linux-tablet-mode script from scresante but I have an issue (https://github.com/scresante/linux-tablet-mode/issues/5).

RussianNeuroMancer, 2019-07-05:
For auto-rotation try Gnome Shell Wayland session on Ubuntu 19.04 with updates installed from proposed (there was changes in libmutter 3.32.2 that should make touchscreen autorotation works on mode devices).